don/cordova-plugin-ble-central #548: Can we use this plugin to communicate between 2 android phones ?

Shery11, 2018-04-07:
I am trying to develop a chat app using this plugin in ionic. 

Let me know

don, 2018-04-19:
This plugin allows you to connect to a Bluetooth LE peripheral.  In order to communicate with another Android phone, you'd need different code to create the peripheral.

I have another plugin [cordova-plugin-ble-peripheral](https://github.com/don/cordova-plugin-ble-peripheral), but unfortunately I haven't implemented it for Android yet. There may be other cordova plugins that let you create peripherals on Android.

closed 2018-04-19 by don

don, 2018-05-25:
@Shery11 [cordova-plugin-ble-peripheral](https://github.com/don/cordova-plugin-ble-peripheral) now has support for Android.

Shery11, 2018-05-25:
Alright. So now it does provide support for communication between 2 android phones using BLE ? right ?

don, 2018-06-07:
@Shery11 yes, but not out of the box. You need to advertise a service on one phone and use this plugin on a 2nd phone. You need to define a Bluetooth service or reuse a service like Nordic UART service. The [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) in this project should be able to talk to the [uart-example](https://github.com/don/cordova-plugin-ble-peripheral/tree/master/examples/uart) in the [cordova-plugin-ble-peripheral](https://github.com/don/cordova-plugin-ble-peripheral) project.