don/cordova-plugin-ble-central #549: ionic ble How to search for wireless scan code gun?

ZGarvey, 2018-04-13:
Hi,
I'm using the BLE plugin, and facing such an issue (Android):

I use the scan () can search to the MAC computer bluetooth, wireless bluetooth mouse and keyboard.But search less than wireless bluetooth scan code gun how to return a responsibility?

Sweep code gun models: symbol CS4070

don, 2018-04-19:
That scanner doesn't support Bluetooth Low Energy. According to the [spec sheet](https://www.zebra.com/content/dam/zebra_new_ia/en-us/solutions-verticals/product/barcode-scanners/Companion/Symbol%20CS4070%20Companion%20Scanner/spec-sheet/cs4070-specification-sheet-A4.pdf) it only supports Bluetooth Classic.

Since it supports Serial Port Profile (SPP) you might be able to use https://github.com/don/BluetoothSerial.

closed 2018-04-19 by don

ZGarvey, 2018-04-20:
@don Ok. thank you very much for your reminding. thank you