don/cordova-plugin-ble-central #552: Get notification state. Only subscribe once.

Domvel, 2018-04-23:
Is there a way to get the notification state of a characteristic? My issue is, that it's possible to subscribe to a characteristic multiple times. In my opinion, it should always be subscribable only once.

If I try stop Notification before start Notification (in then of promise), the subscription does not work.
The start notification method will reject it in this case.

Request:
startNotification() should subscribe only once each service-characteristic.
Or implement a method to check the notification. "isNotificationSubscribed".

don, 2018-04-23:
Currently there's no way to do this. You'd need to read the 2902 descriptor to determine the notification state.

"startNotification() should subscribe only once each service-characteristic." I think this is application's responsibility.

If Live Reload is causing problems, maybe you can add some hooks to disconnect the device, or record the current state. In my workshops that use PhoneGap Developer App, I added hooks to disconnect the device when reloading the app. See https://github.com/don/phonegap-ble-workshop/blob/master/connect/www/js/disconnector.js

Having a isNotificationSubscribed or getNotificationState function is a good idea.

Domvel, 2018-04-23:
Thanks @don for the idea. It works. In a bit other way (because ionic, angular, typescript, etc). But it works.

Assimilate, 2020-08-31:
@Domvel, how did you resolve the issue of not being able to subscribe to notifications once you've stopped one? This seems to apply even if i stop an unrelated notification characteristic. Does this error only show up when using promises, if that's the case, why do you think it is that way?