don/cordova-plugin-ble-central #555: startStateNotifications on iOS not working if connected to device

jake-freshinup, 2018-04-25:
I have a function that alerts me when bluetooth status changes, using startStateNotifications

Before I connect to a device, I can toggle my devices bluetooth on/off and the alert always fires showing me the current state.

Once I connect to a bluetooth device, if I disable my bluetooth I do not get any further notifications.  Also, I had placed an alert in the failure callback during connection and that doesnt seem to fire either.

don, 2018-04-25:
I can not duplicate this. Try running this project https://github.com/don/ble-state-notifications/tree/issue-555 On my iPhone 7 with iOS 11.3 the app updates the Bluetooth state even when connected to a peripheral.

closed 2018-05-07 by don

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios