don/cordova-plugin-ble-central #557: Ionic 3/4 plugin

soesantos, 2018-05-02:
Hi bro,
Do have plan make plugin for support ionic 3/4 plugin....?

Very hard to implement to ionic 3/4 print qrcode now...

Thank you very much
GBU

don, 2018-05-02:
@soesantos this plugin works with Ionic. The Ionic native call this plugin. https://ionicframework.com/docs/native/ble/

I have Ionic examples https://github.com/don/ionic-ble-examples that have been written with Ionic 3. They should work with Ionic 4.

closed 2018-05-02 by don

soesantos, 2018-05-02:
hi don,
i try to print QR code generated to ionic card div (red mark) this is Base64 String of Image  ,

i send code "write" to printer 
"
let PrintData = document.getElementById("SKU")
bluetoothSerial.write(PrintData, success, failure);
"
but the output on the printer is print the string of base64 image not print qr code picture
eg. 
[image]
[image]

don, 2018-05-04:
@soesantos the printer is printing the encoded image. You probably need to send a command code to have it print images. You might need a printer specific plugin like https://github.com/giorgiofellipe/cordova-plugin-datecs-printer

laaaaanceeee, 2018-08-21:
Would you create a peripheral one?

alejandrojcc, 2019-05-11:
Hi @soesantos, can you print qrcode?