don/cordova-plugin-ble-central #558: Error building plugin for iOS with Adobe PhoneGap Builder

fmgomes, 2018-05-02:
Hi, I'm building a app with ble-central plugin, locally using Cordova for android and using Adobe Builder for iOS and Android. The android build works, but for iOS I get an 'error 65'. Any idea of what could be the problem? Is anyone using this plugin on iOS apps built with Adobe Builder? I'm using now version 1.1.8 but tested also with 1.1.4 and the result was the same.

Best regards

Fernando

The following build commands failed:
	CompileC /Library/Developer/Xcode/DerivedData/woke-dgslylmzkdaiohdzjsupdmmtgqva/Build/Intermediates.noindex/ArchiveIntermediates/woke/IntermediateBuildFilesPath/woke.build/Release-iphoneos/woke.build/Objects-normal/armv7/BLECentralPlugin.o woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m normal armv7 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Error code 65 for command: xcodebuild with args: -xcconfig,/private/project/cordova/build-release.xcconfig,-workspace,woke.xcworkspace,-scheme,woke,-configuration,Release,-destination,generic/platform=iOS,-archivePath,woke.xcarchive,archive,CONFIGURATION_BUILD_DIR=/private/project/build/device,SHARED_PRECOMPS_DIR=/private/project/build/sharedpch

nrobertus, 2018-05-02:
Possibly related, but now when I build app with Adobe Phonegap's web app, I get the message "Error - The following 3rd-party plugin is causing the build to fail and may need to be updated to a newer version: cordova-plugin-ble-central." This was the case for both 1.1.4 and 1.1.8, and just recently showed up. Prior to this, I had no issues and had built for iOS hundreds of times.

don, 2018-05-02:
@fmgomes @nrobertus what version of PhoneGap are you using?

joubertje, 2018-05-03:
Good Day,

I am also having trouble with the same error as from yesterday.

"Error - The following 3rd-party plugin is causing the build to fail and may need to be updated to a newer version: cordova-plugin-ble-central"

Ive tried cordova  versions 6.1.0, 6.5.0 and 7.1.0 

Here is the error in the log file:

** BUILD FAILED **

The following build commands failed:
	CompileC build/NZFLT.build/Release-iphoneos/NZFLT.build/Objects-normal/armv7/BLECentralPlugin.o NZFLT/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m normal armv7 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)

I hope this helps.

Justin Joubert

LarsWiegers, 2018-05-03:
@don you might want to look at the variable `discoverPeripherialCallbackId `which is in 
`/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m:267:5: error: use of undeclared identifier 'discoverPeripherialCallbackId'; did you mean 'discoverPeripheralCallbackId'?`.

I believe you mis wrote the variable / identifier name. 

Which is file /src/ios/BLECentralPlugin.m and was changed in commit 8c01fbf4ea8bca436ff8b3133ebc5e8469608c70

If you want i can provide a bigger error message that phonegap gives in the log tab of my project which shows the same error code as the guys above did.

fmgomes, 2018-05-03:
I confirm that previous in the log I pasted above I have the same situation reported by @joubertje. I'm using Cordova 1.6.5 to build locally in my PC only for Android and I am using Adobe PhoneGap Builder (online) to build for both Android and iOS, and when I started using the Adobe Builder I found this problem. I have tried forcing the ble-central version 1.1.4 (the one I was using locally only for Android) and the last one (1.1.8) and the result for iOS using the Adobe Builder was similar. 

/project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m:267:5: error: use of undeclared identifier 'discoverPeripherialCallbackId'; did you mean 'discoverPeripheralCallbackId'?
    discoverPeripherialCallbackId = [command.callbackId copy];
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    discoverPeripheralCallbackId
In file included from /project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m:19:
/project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.h:28:15: note: 'discoverPeripheralCallbackId' declared here
    NSString* discoverPeripheralCallbackId;
              ^
/project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m:281:5: error: use of undeclared identifier 'discoverPeripherialCallbackId'; did you mean 'discoverPeripheralCallbackId'?
    discoverPeripherialCallbackId = [command.callbackId copy];
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    discoverPeripheralCallbackId
In file included from /project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.m:19:
/project/woke/Plugins/cordova-plugin-ble-central/BLECentralPlugin/BLECentralPlugin.h:28:15: note: 'discoverPeripheralCallbackId' declared here
    NSString* discoverPeripheralCallbackId;
              ^
2 errors generated.

joubertje, 2018-05-03:
HI All,

I manged to build by forcing the ble_central version back to 1.1.4 using in the following line without the ~ in front of the version.

	**_<gap:plugin name="cordova-plugin-ble-central" source="npm" spec="1.1.4" />_**

the ~will force it use the latest version.

I also used cordova 6.5.0.

Regards Justin Joubert

don, 2018-05-03:
@fmgomes there was a spelling error in the code discoverPeripherialCallbackId vs discoverPeripheralCallbackId I missed when resolving a merge conflict. Should be fixed now. Try 1.1.9

closed 2018-05-03 by don

fmgomes, 2018-05-03:
@don it now builds without any error, I'll try the app later.

Thanks for your effort!

Best regards

Fernando

joubertje, 2018-05-03:
Thank you @don! Working like a charm again!

commit 485eeca6 (2018-05-03): fix variable name spelling from bad merge resolve
fixes #558