don/cordova-plugin-ble-central #559: ble.write function is not working for the scanned and connected device on iOS

MunkhJukov, 2018-05-04:
Hi dev Team.
How are you?
I just used your library for ionic BLE communication app on iOS and Android.
The Android version working well for scan, read and write.
But for the iOS version, read function working well, but not for iOS.
Please let me know what I have to do now.
Many thanks.
Munkh.

don, 2018-05-07:
@MunkhJukov What error are you getting? Can you create a very simple example to demonstrate the problem?

MunkhJukov, 2018-05-07:
Yes, I can share with you the sample project about BLE error.
https://github.com/MunkhJukov/ble_att.git
Please check this project and I am sure you can see the plugin not working error.
Thanks.

MunkhJukov, 2018-05-08:
Anyone can help me PLZ?

don, 2018-05-10:
@MunkhJukov can you give me more details about what's not working? How do you know it's not working? Are you getting an error message? Is there a stack track in JavaScript? Is there a stack trace in the native code? I'm not seeing `ble.read` in [your code](https://github.com/MunkhJukov/ble_att.git). The more details you can provide, the easier it is to track down problems.