don/cordova-plugin-ble-central #563: Write more then 20 bytes. Read works.

Domvel, 2018-05-07:
I have a BLE server (periphery) with a characteristic more than 20 byte data (about 200). 
if I use the write method of this ble central library I get an exception error "14" (as number). 
(Error 14 (0xE) GATT ERR UNLIKELY)

Reading works with more than 20 bytes, but writing fails. Is this generally not supported?

don, 2018-05-07:
@Domvel in my experience, it depends on the peripheral. Both Android and iOS use BLOB write to write characteristics larger than 20 bytes. For some peripherals I need to break writes up into smaller chunks. Can you write 180 bytes or 100 bytes? For some peripherals we've had to break into 20 byte chunks.

Since you're getting (Error 14 (0xE) GATT ERR UNLIKELY) it's possible there's an error on the peripheral side?

If you're using Android, try [enabling the btsnoop_hci.log](http://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) to see if there more info over the wire.

Domvel, 2018-05-07:
@don Thanks for the fast response. :) 
It fails on 21 bytes and more. I also tried several bluetooth apps from the play store. 
nRF Connect also returns error 14. BLE Scanner split the message into 20 byte messages. (without relation. For the peripherial it looks like 2 separated write requests.)

How to break it into 20 byte chunks? I need an offset. Otherwise the peri. write from offset 0 each chunk.
I do not know which side has the problem. What does the error 14 Unlikely mean? ^^

I've enabled the bt snoop hci log and pulled with `adb pull /sdcard/btsnoop_hci.log` and viewed in Wireshark. But I've no idea what the log want to tell me. 😄 

I'll try this with another peripherals to exclude the device. But you say that this library can write more than 20 bytes for sure?

don, 2018-05-07:
See this example https://gist.github.com/don/ad3d6f444925c57d169baf54e262348f for one way to break data into chunks and send it.

Domvel, 2018-05-07:
Oh, right. I already tried it. The periphial has to handle the data by a callback and write it to a buffer. In this case you have to send a begin and commit command that the periphial can reset the buffer and know when the writing is done.

I don't understand completely. Why is you example / case limited to 500 bytes?
A writing of a chunks is equal as another write with complete data (<= 20 byte)
If you sent large data as chunks, the receiver has to be handle it and add received data to a (e.g.) lokal buffer.
So you have to ensure that the messages are sent in the correct order. And tell the server when the buffer starts and ends.
You can do it by writing into another chacteristic like "I'll start to send chunks" and "I'm done.". Or encode it to the chunks you sent with a custom protocol.

Boring background in my case:
I've to send about 500 KB over BLE. Writing without response brought us an acceptable speed but some non catchable writing issues. And writing with response the speed breaks down to factor 10 times slower, but without problems. With data handling of the application on the periphial it takes 100 minutes. Without response about 10 minutes. (a software update for the device, downloaded via the app). So we wanted to write more data per message with response to speed it up without having writing problems. Whatever ... In my case I have two options for now: 

- Send the data without response by 20 byte chunks and find out the writing issues.

- Find out the problem of writing with more than 20 bytes.

Just to be sure, this BLE plugin can send more than 20 bytes, right? Already tested?
So I can exclude a client side issue. Anyway I'll check it with another periphial. 

Thanks for your support and sharing your experience.
I'll write back after tested some cases. If your sure that this library can write more than 20 bytes on Android, you can close this issue.
I just thought this is an issue and may be suggested.

Domvel, 2018-05-08:
@don If I request a higher MTU size, the writing works. But I can't request the MTU size with this plugin, right? There is no API function. I've tested it with the App nRF Connect on Android.

Domvel, 2018-05-08:
I found (this)[https://github.com/don/cordova-plugin-ble-central/issues/234] issue with similar topic. And a suggestion of a method to request the MTU size. But I have no idea / experience how to include and use it. 
Could you include this feature?

Btw. it seems to be a Android problem. I tested the same with an iOS Device (iPhone 5s, 11.2.5). Also with the same App (nRF Connect). And it workes to send more than 20 bytes.

Domvel, 2018-05-09:
see #567 and pull-request #568

closed 2018-05-09 by Domvel