don/cordova-plugin-ble-central #566: Read Status=133

KamarulAyi, 2018-05-08:
Hello,
I am new in BLE. So far I manage to scan and connect to desired devices by using this plugin in Ionic.
After the device connected, I got this data: 

{"name":"DEVICENAME     ",
"id":"00:A0:50:E1:78:41",
"advertising":{},
"rssi":-43,

"services":[
    "1800",
    "1801",
    "1111"
],
"characteristics":[
    {"service":"1800",
    "characteristic":"2a00",
    "properties":["Read"]},

    {"service":"1111",
    "characteristic":"2112",
    "properties":["Read"]},
    
    {"service":"1111",
    "characteristic":"2114",
    "properties":["Read","WriteWithoutResponse"]},
    
    {"service":"1111",
    "characteristic":"2116",
    "properties":["Read","WriteWithoutResponse"]}
]
}

I already try to read ("00:A0:50:E1:78:41","1800","2a00") and its successfully returned the value which is the device name.
But when I want to read the other services which is as below :

`this.ble.read("00:A0:50:E1:78:41","1111","2112").then(
      r2112 => console.log(JSON.stringify(r2112)) ,
      err => console.log("Error : "+JSON.stringify(err))
    )`

It return this in console : "Error reading 00002112-0000-1000-8000-00805f9b34fb status=133"

Device : Android Lollipop 5.1, Bluetooth 5.1

Please let me know what I should do.
Thank you.

don, 2018-05-10:
@KamarulAyi your code looks OK. Can you see if you can read 2112 using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US)?

Error 133 (0x85) is a generic GATT_ERROR, it doesn't give a lot of information. Did your device disconnect? Did the Bluetooth stack crash? (This happens a lot on older Android versions.) Rebooting the phone sometime fix issues.

KamarulAyi, 2018-05-10:
The characteristics "2112", "2114" and "2116", I can't read using nRF Connect ,it return Error 133 (0x85).

The device stay connected by using isConnected function.
I will check on the Bluetooth stack.

So far I already test and retest (after reboot) by using two different Android devices : 
1. As mentioned
2. Android 7.0
Both give the same result which is fail to read.

I also tried to use other plugin, same result.

Are there any special case that I need to look into regarding Read function?

don, 2018-05-10:
Since nRF Connect, different Android devices and another plugin all give you the error, it sounds like the device is the problem. 

The device could be in a bad state. Maybe you need to write to another characteristic before you can read. Maybe you need to bond to the device?  Can the device manufacturer can provide some sample code or guidance?

KamarulAyi, 2018-05-22:
Thank you so much don, the device need to be paired first because the characteristic was encrypted.
I also got extra documents as a guidance from the manufacturer and it's working fine.

closed 2018-05-22 by KamarulAyi