don/cordova-plugin-ble-central #567: Request MTU Size

Domvel, 2018-05-09:
Feature request for API method `requestMtu` [reference](https://github.com/develop-algoritma/cordova-plugin-ble-central#requestmtu)

This is an Android issue. For larger characteristics (>20 bytes) it is required that the developer request a higher MTU size. A write request must complete in one message. (MTU size > Characteristic Size + Meta)

### requestMtu

```js
ble.requestMtu(deviceId: string, mtuSize: number, [success], [failure]);
```

On iOS this is not required. It's a Android only feature for this plugin.

Domvel, 2018-05-09:
@don Please merge and release. :)

Domvel, 2018-05-09:
# Use Case

You want to send more than 20 bytes to a characteristic with more than 20 byte size? On Android?
Maybe this doesn't work and fails with Error 14 "Unlikely". 

Known issue: The data to write has to be match in one message. So the MTU size has to greater than the data to send / characteristic size. 
The default MTU size is 23 (20 bytes payload + 3 byte header). But the meta data length depends on different things. Whatever ... a easy calculation: Use the max. MTU size (e.g. 512) or your characteristic size + a hand full bytes. (For more information about this topic, ask the internet.) A exact MTU size does not matter in our (my) case.

Now you can call the MTU method after device connection like: 

```js
ble.requestMtu(this.device, 512, () => {
  console.log('MTU Size ok.');
}, error => {
  console.log('MTU Size failed.');
});
```

That's it. Happy writing.

don, 2018-05-10:
Thanks for the pull request. I'll test and merge this week.

closed 2018-05-10 by don

tempestwf, 2018-05-25:
Hi

We are in need of this functionality. Any idea when it will be packaged into a release so we can point our package.json at it?

Thanks very much.

Will

don, 2018-06-07:
@tempestwf planning on a release this week

I-Connect, 2018-10-25:
Not sure if it is (always) needed but I have found a delay/sleep of 100 ms is needed within my app after requestMTU before executing other BLE reads/writes. Maybe only for larger messages...?

Also in general between other reads and writes? I have been struggling with this for multiple hours now and I found that using a sleep of 10ms after each BLE action resolves my issues.
This probably prevents BLE stack congestion...? 

Can you confirm if this is needed?

vallettea, 2020-11-04:
I also needed to wait 100ms after MTU change otherwise no packets could be transfered