don/cordova-plugin-ble-central #569: Devices are not found when running ble.startScan (on some phones)

nimeso, 2018-05-12:
On some phones the BLE devices are not being detected. I'm a bit new to BLE so bare with me if I'm just a noob.
Phone: Vodafone Smart N8, running Android 7.0
framework: Ionic 3
it's working fine on my phone so I'm not sure what is going on... any ideas would be greatly appreciated. 

scanForDevices(){
       console.log("scan for devices");
       this.ble.startScan([]).subscribe(device => {
        if (device.name.indexOf('Sensibel') == -1) {
          return
        } else {
           if(device.id == localStorage.getItem('buttonID')){
            console.log("Device found - "+device.id );
            this.zone.run(() => {
              this.deviceList.push(device);
            });
            this.connect(device);
          }else{
            this.zone.run(() => {
              this.deviceList.push(device);
              this.showWizard = true;
            });
          }
        }
      });
    }

don, 2018-05-12:
Make sure Bluetooth is on. On Android make sure location services are on. When a phone is connected to a peripheral, most peripherals stop advertising, so other phones won't see them.

Try scanning with LightBlue Explorer or nRF Connect. Do you see different results than your app?

nimeso, 2018-05-12:
Thanks heaps Don.

Yes nRF shows them all. My app does not.
I'm using zones to refresh the list and everything so I'm really not sure
whats going on.
I've got to go out for a bit but I'll update this issue after I can do some
more testing.
Does my code look ok?

Cheers

don, 2018-05-12:
Nothing jumps out in the code. If possible I like to limit the devices I scan for using a service_uuid, but that's not always possible. Maybe log the device id and name for device not named Sensibel. The scan doesn't always get the name, especially if the scan response has different info.

Domvel, 2018-05-29:
This has to do with #527 This plugin should handle the location requestion automatically. On Android (>=23?) a permission dialog will appear if the app request location authorization. Alternative use another plugin to do this e.g. https://github.com/dpa99c/cordova-diagnostic-plugin

don, 2018-06-08:
Assuming you've figure this out by now or it has to do with permissions which now give better error messages.

closed 2018-06-08 by don