don/cordova-plugin-ble-central #570: Attempt to invoke virtual method on a null object reference.

Domvel, 2018-05-14:
Sometimes I get the error:

```text
"Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference"
```

I have successfully connected and want to subscribe to a notification of a characteristic that exists.
I wondering what can produce this error? Is this a known issue?

don, 2018-05-14:
Can you reproduce this? Can you get the stacktrace from `adb logcat`? It sounds like the service is null somewhere. Maybe I'm calling getCharacteristics before the service had been discovered?

Domvel, 2018-05-15:
The periphery has lost all (custom) services / characteristics. -.- Don't know why. Not a fault of the plugin. Sorry. BUT maybe the error message should be improved / error catched. 

Btw. I have the following issue:
If I want to subscribe to a non-existing characteristic, no other notifications will work anymore.
Case: 
1. Subscribe to a existing characteristic. (works)
2. Subscribe to a non-existing characteristic. (will terminate all notficiations. including of working characteristics)

Maybe this issue (the error message) has to do with.

Here's the `logcat`

```text
05-15 10:30:09.990 20376 20376 I chromium: [INFO:CONSOLE(156)] "Start Notify", source: http://192.168.1.42:8100/__ion-dev-server/ion-dev.js?v=3.1.9 (156)
05-15 10:30:09.991 20376 20665 E PluginManager: Uncaught exception from plugin
05-15 10:30:09.991 20376 20665 E PluginManager: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.findNotifyCharacteristic(Peripheral.java:478)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.registerNotifyCallback(Peripheral.java:391)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.processCommands(Peripheral.java:769)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.queueCommand(Peripheral.java:735)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.Peripheral.queueRegisterNotifyCallback(Peripheral.java:671)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.BLECentralPlugin.registerNotifyCallback(BLECentralPlugin.java:474)
05-15 10:30:09.991 20376 20665 E PluginManager:         at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:217)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
05-15 10:30:09.991 20376 20665 E PluginManager:         at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:9)
05-15 10:30:09.991 20376 20665 E PluginManager:         at android.os.Handler.dispatchMessage(Handler.java:105)
05-15 10:30:09.991 20376 20665 E PluginManager:         at android.os.Looper.loop(Looper.java:156)
05-15 10:30:09.991 20376 20665 E PluginManager:         at android.os.HandlerThread.run(HandlerThread.java:61)
05-15 10:30:10.011 20376 20376 I chromium: [INFO:CONSOLE(156)] ""Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference"", source: http://192.168.1.42:8100/__ion-dev-server/ion-dev.js?v=3.1.9 (156)
```

Domvel, 2018-06-04:
New case on invoking the `disconnect` method:
```text
Runtime Error: Uncaught (in promise): Attempt to invoke virtual method `void android.BluetoothGatt.close()` on a null object reference.
```

On retry, the disconnection was successfully. But only for the app. The peripheral was still connected.
Weird. I can't explain the reason. Some ideas?

don, 2018-06-07:
The Android disconnect logic should be a lot better because of @doug-a-brunner's changes which are included in 1.2.0. Closing as fixed. Re-open if you continue to see this.

closed 2018-06-07 by don