don/cordova-plugin-ble-central #573: Unable to start scan with multiples service UUIDs

lulumahot, 2018-05-17:
I try to start a scan on two service UUIDs on Android but the callback is never called.

If try with only one UUID at a time it works fine.

Are you able to reproduce it ?

don, 2018-05-17:
This is an Android limitation. Android is looking for a device that has both UUID1 *and* UUID2.

I disagree with Android's implementation, but there's not way to change this. I think you need to scan once for each UUID.

lulumahot, 2018-05-18:
I see you're using the old Android Bluetooth API and I know this is possible using the new one. Is there any reason you're using the deprecated API (other than compatibility) ?

don, 2018-05-18:
> Is there any reason you're using the deprecated API?

Mostly because the plugin is has been around for a while and Cordova runs on really old Android. At some point I'll upgrade once Cordova drops KitKat support.

closed 2018-06-07 by don