don/cordova-plugin-ble-central #574: Devices with Pin Requirements

brassier, 2018-05-18:
I love the plugin so far.  I'm trying to use this plugin to connect to a printer which requires a PIN to connect.  

When connecting to the device with the plugin I am getting a successful connection, but when going into the Bluetooth settings the device is not actually connected.  It does seem semi-connected, because when I turn the printer off the disconnect event does get triggered.  I am also able to call `readRSSI()` on the device.   However, after connecting, the plugin is in an odd state where it doesn't really see the device anymore until I re-start the app.  For example, if I scan again, the device I connected to never appears.  Then after restarting the plugin it does see the device as not connected again.

When manually connecting to this device via Bluetooth settings, it gives an error stating that it needs a PIN.  I'm wondering why this plugin isn't responding with an error like that?  Is there a way to enter a PIN during the connection to get a complete connection?

don, 2018-05-18:
Most BLE devices don't require PINs. If the device wants to be bonded, sometimes you need to enter a pin. After you connect, if you read or write to a characteristic that has permissions requiring encryption, you should be prompted to enter a pin. The plugin does not have explicit support for bonding yet. You might need to bond the device outside of the app. You can try using nRF Connect or Light Blue Explorer.

brassier, 2018-05-18:
Thanks for the response @don!  One other thing I just noticed is if I pair (bond?) outside of the plugin using the PIN, the device shows as paired in Bluetooth settings.  However, when calling `isConnected()` with the plugin, it is throwing an exception, which as I understand it is supposed to indicate it is not connected.  Do I have that wrong?  Or maybe `isConnected()` will not act normally with these PIN devices?

don, 2018-05-18:
After you bond with an external app, disconnect the peripheral. Then in your app, scan and connect to the peripheral. Right now, the plugin doesn't have an API for Android's [getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#getBondedDevices()). iOS doesn't have an equivalent API and would need wrappers for [retrieveConnectedPeripherals(withServices:)](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipherals) or [retrievePeripherals(withIdentifiers:)](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1519127-retrieveperipherals).

brassier, 2018-05-21:
Hi @don - thanks again for the response.  I'm trying to find alternative ways to connect my app to this printer.  I noticed you are also the owner of the [BluetoothSerial](https://github.com/don/BluetoothSerial) plugin.  I see that has some support for bonding.  Does that plugin have any ability to handle PINs during bonding?

don, 2018-05-21:
IIRC the bonded reference in BluetoothSerial refers to BluetoothClassic not BLE.

don, 2018-06-13:
@brassier There's still no PIN or bonding support, but once you've bonded in the Android settings, you can use [ble.bondedDevices()](https://github.com/don/cordova-plugin-ble-central#bondeddevices) to get your device then connect. Alternately if you know the MAC address, can just call `ble.connect(MAC, success, failure)` without scanning. `ble.isConnected` shouldn't give you a stack trace. If you can, open a new issue with the stack trace and steps to duplicate so I can fix.

don, 2018-06-13:
Created #605 for createBond

closed 2018-06-13 by don