don/cordova-plugin-ble-central #576: Could not find peripheral (null).

rooz121359, 2018-05-21:
I sometimes get this error when i try to connect to a bluefruit device!

`Could not find peripheral (null).`

The issue is that it doesn't happen all the time and when it happens, if i keep trying to connect, after a 3 or 4 attempts, it will connect without any issue!

Is this a known problem?

any suggestions would be appreciated.

don, 2018-05-21:
@rooz121359 on iOS or Android?

rooz121359, 2018-05-21:
on iOS don.

don, 2018-05-29:
@rooz121359 I think your JavaScript code is passing `null` to `ble.connect`. 

The error is from here. That error message comes from https://github.com/don/cordova-plugin-ble-central/blob/82a93ba9022f298b71db7f066ec34331a8e610e7/src/ios/BLECentralPlugin.m#L75 because the peripheral is null.

The UUID is here, which is passed to the ObjectiveC from JavaScript https://github.com/don/cordova-plugin-ble-central/blob/82a93ba9022f298b71db7f066ec34331a8e610e7/src/ios/BLECentralPlugin.m#L65

closed 2018-06-07 by don