don/cordova-plugin-ble-central #577: iOS - startNotification callback not raised after disconnection from the peripheral

zabojad, 2018-05-25:
Hi !

I'm having this issue only on iOS with `cordova-plugin-ble-central` 1.1.9.

I'm connecting to a peripheral and listens to its notifications. Everything works fine.

Now, if I:
- disconnect from the iOS app and reconnect to the peripheral, everything is OK, I start notifications and receive them.
- turn off the peripheral and turn it on again. The app is able to find the peripheral, connect to it but the start notification callback will never be raised.

Can you please fix it?

zabojad, 2018-05-28:
Hi @don ! Any thought on this one?

don, 2018-05-29:
@zabojad I can't duplicate this

I'm using this project https://github.com/don/ITP-BluetoothLE/tree/master/phonegap/thermometer

    cordova platform add ios
    cordova plugin add cordova-plugin-ble-central
    cordova plugin add cordova-plugin-dialogs

Run the app, connect to the thermometer peripheral. After a few notifications, pull the power on the peripheral. App gets the disconnect dialog then goes back to scan screen after clicking ok. Turn the peripheral back on. Connect in the app. I see notifications.

Do you see the same behavior if you use [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8)?

Can you run a BLE sniffer to see if the peripheral if the 2902 descriptor is being set and the peripheral is sending a notification after the reconnect?

zabojad, 2018-06-07:
Hey @don !

I do not reproduce the issue with LightBlue Explorer (nor with cordova-plugin-ble-central on Android).

> Can you run a BLE sniffer to see if the peripheral if the 2902 descriptor is being set and the peripheral is sending a notification after the reconnect?

zabojad, 2018-06-07:
I'm trying to debug the problem and for a reason I do not understand, `didUpdateValueForCharacteristic` is never raised when the problem occurs...

zabojad, 2018-06-07:
> Can you run a BLE sniffer to see if the peripheral if the 2902 descriptor is being set and the peripheral is sending a notification after the reconnect?

don, 2018-06-07:
@zabojad For iOS I like using the [Nordic nRF Sniffer](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF-Sniffer) with the nRF51/51 SDK to man-in-the-middle the connection. I think that only works if your peripheral is nRF5x based.

That's odd that `didUpdateValueForCharacteristic` is never called. After you re-connect, are you calling `ble.subscribe` again? Is it possible to make a very simple app that causes the problem?

zabojad, 2018-06-07:
@don 

> After you re-connect, are you calling ble.subscribe again?

You mean (`startnotification`)[https://github.com/don/cordova-plugin-ble-central#startnotification]?

Yes, I call it right in the success callback of the (`connect`)[https://github.com/don/cordova-plugin-ble-central#connect] method...

don, 2018-06-07:
> You mean ble.startnotification?
yes

don, 2018-06-07:
Is this the same problem at #476?

zabojad, 2018-06-07:
No unfortunately, I've tried this fix and it did not sole my issue... In my case, I do not call `ble.stopNotification`...

don, 2018-06-13:
@zabojad Can you try the latest commit ac55995dc9e9d19f231c13e6bbfb9ba24b076480 to see it makes a difference?  If it's still broken, can you share some code with me that causes the issue?

Also what version of iOS and Xcode are you running?

zabojad, 2018-07-03:
@don please let this ticket open, I'm going to vacations and won't be able to work on this until August...

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios

gopaljatiya, 2021-06-18:
Hi , has anyone solved this issue. We are facing the same issue on iOS only, where if we reconnect bluetooth and call startNotification, the subscribe even is never called , i.e. we are not getting any response. It works fine for the first time connection. Any pointers would really help.

peitschie, 2022-02-15:
I've noticed that if the `startNotification` fails during setup on iOS that the error is never passed back to the javascript layer.
I'm not sure why reconnecting would impact this or make it more likely to fail however, unless the peripheral is rejecting the request to activate notifications for some reason.

There's an obvious bug to fix at least, so I'll put up a PR for this.

peitschie, 2022-02-17:
A fix is available in pre-release in case anyone has time to test it: https://www.npmjs.com/package/cordova-plugin-ble-central/v/1.4.3-alpha.1

tigohenryschultz, 2022-02-18:
I tested 1.4.3-alpha.1 on iOS and my startNotification is not returning success scenarios

peitschie, 2022-02-21:
@tigohenryschultz are you able to move that to a new issue... that sounds unrelated to the described bug here.

closed 2022-03-14 by peitschie