don/cordova-plugin-ble-central #579: Reading on interval from a characteristic leads to a disconnect on Android

orlyapps, 2018-05-29:
Hello,

we have discovered a bug on android while reading from a characteristic. 

**Describe the Bug**
After 400-500 reads the android device just disconnects.

**How to Reproduce**

1. We are using your ionic thermometer demo ( https://github.com/don/ionic-ble-examples/tree/master/thermometer )
2. And making the following changes to your onConnected Method
```typescript
  async onConnected(peripheral) {
    this.peripheral = peripheral;
    this.setStatus("Connected to " + (peripheral.name || peripheral.id));

    // Subscribe for notifications when the temperature changes
    this.ble
      .startNotification(this.peripheral.id, THERMOMETER_SERVICE, TEMPERATURE_CHARACTERISTIC)
      .subscribe(
        data => this.onTemperatureChange(data),
        () => this.showAlert("Unexpected Error", "Failed to subscribe for temperature changes")
      );

    let polling = true;
    while (polling === true) {
      try {
        let data = await this.ble.read(this.peripheral.id, THERMOMETER_SERVICE, TEMPERATURE_CHARACTERISTIC);
        console.log(data);
      } catch (error) {
        console.log(error);
        polling = false;
      }
    }
  }
```
3. After 1-2 minutes and 400-500 reads we get this error ```Error reading 0000bbb1-........ status=242```
4. Bluetooth Disconnect is called

**Others observations**

* icordova-plugin-bluetoothle
* Tested on Android 7 & 8 and different devices Nexus 4, Samsung Galaxy Tab 2, Zebra TC25, Nexus 5X
* Adding a delay (3-5 seconds) between read requests dont help. But it takes to fail

orlyapps, 2018-05-30:
Hello Don and thanks for your replay.
I like to add that we have tried two different peripherals. A custom made peripheral and a virtual peripheral with the LightBlue App (https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8)
Both have the same problem.

Also on iOS with your plugin and on Desktop with electron/ noble we cannot reproduce the issue.

don, 2018-06-07:
@orlyapps it sounds like an Android issue with some devices or Android versions? It's odd that you see this on a Nexus 5X but I don't. I'm going to close this issue because I don't think it is a plugin problem. If you find otherwise and there are changes I can make, reopen the issue.

closed 2018-06-07 by don