don/cordova-plugin-ble-central #580: Calling isEnabled on ios brings up a dialog asking the user to enable bluetooth.

cairinmichie, 2018-05-30:
I would much rather control how the dialog looks/works, how do I disable this?

don, 2018-05-30:
That is a problem the dialog is shown for isEnabled and startStateNotifications. It might be possible to skip the check for those two functions. The biggest problem with disabling the dialog is that there's no other legal way to send the user the the Bluetooth settings. Ideally this would be a configuration to enable/disable this. I'll look into this next week when I have more time.

cairinmichie, 2018-05-30:
I created a pull request that shows how to disable it. I then used the [Cordova-open-native-settings](https://github.com/guyromb/Cordova-open-native-settings) plugin to open the iOS settings, and your plugin to open android settings.

don, 2018-06-07:
@cairinmichie do you have iOS apps using [cordova-open-native-settings](https://github.com/guyromb/Cordova-open-native-settings) that have been approved for the store by by Apple?

closed 2018-06-07 by don

cairinmichie, 2018-06-07:
Yes.

don, 2018-06-07:
Add showBluetoothSettings for iOS in #591 based on code from [cordova-open-native-settings](https://github.com/guyromb/Cordova-open-native-settings)

commit a40e5502 (2018-06-07): Merge pull request #584 from cairinmichie/master

Remove dialog for isEnabled and startNotification
Fixes #580