don/cordova-plugin-ble-central #583: BLE is not defined

I-Connect, 2018-05-30:
Hi,

Sorry for raising this again... but I am a bit stuck.
A am working in cordova on a android app.

I removed plugin and android platform.
added android platform 6.2.0
added plugin (v1.1.9) again:

In config.xml:
<plugin name="cordova-plugin-whitelist" spec="^1.3.2" />
    <plugin name="cordova-plugin-ble-central" spec="^1.1.9" />
    <engine name="android" spec="^6.2.0" />

build solution and ran it on a samsung galaxy 8.

In chrome developer tools I get "Uncaught ReferenceError: ble is not defined".

Any idea's?

thx

don, 2018-05-30:
When you're using BLE you must run on a device. An emulator or the browser won't work. You can use chrome developer tools for the code running on your phone by going to chrome://inspect on your computer.

I-Connect, 2018-05-30:
Hi Don,

Thx for the quick reply. I am using a samsung s8 and indeed using chrome developer tools to inspect the code running on my phone.,

don, 2018-05-30:
If you're seeing blue not defined on the phone something is installed incorrectly. I recommend running the latest cordova-android, not 6.2.0. Try these commands below. Backup your project first since it deletes the native Android project.

    cordova plugin rm cordova-plugin-ble-central
    cordova platform rm android
    cordova platform add android@latest
    cordova plugin add cordova-plugin-ble-central
    cordova run android --device

I-Connect, 2018-05-30:
Hi Don,

I used 6.2.0 because I got this error on installing the BLE plugin: "Plugin doesn't support this project's cordova-android version. cordova-android: 7.1.0, failed version requirement:<6.3.0" 

But still, ran the commands you provided, see the output below. It is having trouble installing the latest arduino platform. Giving an error on the instalation of "cordova-plugin-app-preferences".

Building the app now fails.

This error I did not have before. I do not have much experience with cordova, do you see something what could be the cause?

thx,
Jeroen

cordova plugin rm cordova-plugin-ble-central
result:
```
Uninstalling 1 dependent plugins.
Uninstalling cordova-plugin-compat from android
Uninstalling cordova-plugin-ble-central from android
Removing "cordova-plugin-ble-central"
Removing plugin cordova-plugin-ble-central from config.xml file...
Removing cordova-plugin-ble-central from package.json

```
cordova platform rm android
result:
```
some errors about build directories not able to remove, 
Removing platform android from config.xml file...
Removing android from cordova.platforms array in package.json

```
cordova platform rm android (ran second time)
result:
`Removing platform android from config.xml file...`

cordova platform add android@latest
result:
```
on installing apppreferencesactivity.java no such file or directory
Failed to install 'cordova-plugin-app-preferences': Error: ENOENT: no such file
or directory, open 'C:\....\I-Connect\GitHub\C-Sense_app\C-Sense_
app\platforms\android\src\me\apla\cordova\AppPreferencesActivity.java'
```

Ran cordova platform rm android again

Ran cordova platform add android@latest
Same result as above

cordova plugin add cordova-plugin-ble-central
result:

```
**Plugin doesn't support this project's cordova-android version. cordova-android:
7.1.0, failed version requirement:
      <6.3.0**
Skipping 'cordova-plugin-compat' for android
Android Studio project detected
Adding cordova-plugin-ble-central to package.json
Saved plugin info for "cordova-plugin-ble-central" to config.xml
```

cordova run android --device
result:
```
Android Studio project detected
will push strings array {"name":"lang","titles":["English (US)","English (UK)"],
"values":["en-us","en-gb"]}
unhandled exception { Error: ENOENT: no such file or directory, mkdir 'C:\.....\GitHub\C-Sense_app\C-Sense_app\platforms\android\res\xml
'
  errno: -4058,
  code: 'ENOENT',
  syscall: 'mkdir',
  path: 'C:\\......\\GitHub\\C-Sense_app\\C-Sense_app\
\platforms\\android\\res\\xml' }
(node:10124) UnhandledPromiseRejectionWarning: Unhandled promise rejection (reje
ction id: 1): Error: ENOENT: no such file or directory, mkdir 'C:\.....\GitHub\C-Sense_app\C-Sense_app\platforms\android\res\xml'
(node:10124) [DEP0018] DeprecationWarning: Unhandled promise rejections are depr
ecated. In the future, promise rejections that are not handled will terminate th
e Node.js process with a non-zero exit code.
Android Studio project detected
will push strings array {"name":"lang","titles":["English (US)","English (UK)"],
"values":["en-us","en-gb"]}
unhandled exception { Error: ENOENT: no such file or directory, mkdir 'C:\Users\
NL24544\Qsync\I-Connect\GitHub\C-Sense_app\C-Sense_app\platforms\android\res\xml
'
  errno: -4058,
  code: 'ENOENT',
  syscall: 'mkdir',
  path: 'C:\\Users\\NL24544\\Qsync\\I-Connect\\GitHub\\C-Sense_app\\C-Sense_app\
\platforms\\android\\res\\xml' }
(node:10124) UnhandledPromiseRejectionWarning: Unhandled promise rejection (reje
ction id: 1): Error: ENOENT: no such file or directory, mkdir 'C:\.....\GitHub\C-Sense_app\C-Sense_app\platforms\android\res\xml'
(node:10124) [DEP0018] DeprecationWarning: Unhandled promise rejections are depr
ecated. In the future, promise rejections that are not handled will terminate th
e Node.js process with a non-zero exit code.
```
cordova build
result:
```
.....
unhandled exception { Error: ENOENT: no such file or directory, mkdir 'C:\.....\GitHub\C-Sense_app\C-Sense_app\platforms\android\res\xml
.....
```

don, 2018-05-30:
“Plugin doesn't support this project's cordova-android version. cordova-android:
7.1.0, failed version requirement:
      <6.3.0”

Ignore that error, it’s about a dependent plugin that’s not needed for the latest Android. 

I’m not sure what the other problem is. I’d make sure you’re running the latest Cordova on Node 8.x.

Can you see if you can build and run this simple project? https://github.com/don/ble-scan

I-Connect, 2018-05-31:
well.... Long story, got it working and got a whole lot smarter :-)

I updated node and cordova, same plugin path issue.
Removed the cordova-plugin-app-preferences plugin, platform install path issue is now solved

cordova run android --device gave me the following error:
C:\WINDOWS\system32\adb' is not recognized as an internal or external command

Tried to solve this setting path and setting env variables but did not succeed. Starting the ADB server manually solved the issue (as a work around)

But still ->Uncaught ReferenceError: ble is not defined

Build the BLE-Scan app, worked immediately. Comparing the code I found:
 I had:
`document.addEventListener('deviceready', this.onDeviceReady, false);`

Changing it to:
`document.addEventListener('deviceready', this.onDeviceReady.bind(this), false);`

I know how this creeped in, I started with the robosmart example, there the bind is splitted over two functions, I (as a newbee) rewrote/reused this incorrectly and overlooked it when checking it against another app I did have working.

So it was my own code after all...
Thx for pointing me in the right direction!!

closed 2018-06-07 by don