don/cordova-plugin-ble-central #585: The specified device has disconnected from us

edwardpress, 2018-06-03:
I attempted to use this Cordova plugin on iOS. After confirming successful connection, i always get the " the device has disconnected from us" msg whenever I try to send a write request to the Bluetooth device. The same behavior doesn't happen on Android.

I would like to know how to start

edwardpress, 2018-06-06:
After a few testing, it seems like the problem happens because the data i sent to write is too big (bytelength > 1000) . The issue only happens in ios for now

don, 2018-06-06:
@edwardpress try breaking the message into multiple chunks of less than 500 bytes each

Domvel, 2018-06-06:
@edwardpress On android, there is a new method to request the MTU size. This is necessary to write on characteristcs with more than 20 byte. But maximum of 512 byte. 
On iOS the MTU size is handled automatically. BUT in your case you exceed the maximum size of a characteristic / package.

What's the size of your characteristc? 20 byte is the default value. MTU size is by default 23 byte. (default 20 byte of the charactersitc + 3 byte header of a package).
At the paper, a maximum MTU size of 517 is possible. But in my experience, 512 byte is the limit.
Your characteristic should be a few bytes below the chunk size. For example 500 byte. 
To transmit your data (> 1000 byte) you have to split it to chunks.

edwardpress, 2018-06-06:
Here's the funny thing... in Android, it works exactly fine for the data that i send... (At least for android 6+), but for iOS 11, the connection always drop the moment the moment i send the write request (and i only notice the issue after rumming through the thread).

By sending through chunk, I assume that you need to slice the arrayBuffer into an array of arrayBuffer...and then proceed to send a write request for every arrayBuffer...

edwardpress, 2018-06-06:
I assume it could be the device that do the hard work (the device is a bluetooth printer which has multiple characteristics uuid read & write), so i assume the hardware handles it internally when you send the data from android...

For IOS, maybe the device doesn't work as such...but there's no way for me to know. I will update you guys later.

That's for the help :)

don, 2018-06-06:
@edwardpress like @Domvel says the max characteristic size for Bluetooth 4.x is approx 500 bytes. Bluetooth 5.x is 2000 bytes. It's possible that Android is breaking this up for you and iOS is not. You'd need to run a sniffer to be sure what's happening over the wire. Breaking the data into smaller chunks and doing multiple writes is probably and easier solution. 

@Domvel you don't always need to increase the MTU size on Android. Many android devices will transparently do BLOB writes to send larger data.

Domvel, 2018-06-07:
On my project I have one peripheral ble chip and many android devices tested. From android 5.1 to 8.0. From bluetooth 4.0 to 5.0. All devices need to increase the MTU size to a value that a writing matches one message. On read it does not matter. Also on MTU size 23 I can read huge data. Only the writing fails. Maybe a issue of the peripheral? I don't know. This is my first ble project. The peripheral has bluetooth 4.2.

Domvel, 2018-06-07:
@edwardpress Btw. sending huge data as chunk in a cutom way is only possible if you have access to the printer software source code. Otherwise read the description of the printer how to send the data.

edwardpress, 2018-06-07:
Tested with a different android device, and this time i will need to chunk the code to below 256 byte limits for the write operation to succeed. Kinda learn it the hard way that it's device dependant and platform dependant as well.

I think it's better for me to close the issue now that it's not an issue with the ble plugin itself. Thanks for the help and clarification

closed 2018-06-07 by edwardpress