don/cordova-plugin-ble-central #586: Scan should emit complete-callback

Domvel, 2018-06-04:
The `scan` method should have a complete callback. This saves a separate timeout implementation.
Current workaround: Add e.g. `setTimeout(done, time)`; on a invoke of `scan` .

In ionic the complete event is not fired for the rxjs observable. I expected:
```js
this.ble.scan([], 3).subscribe(device => {
  console.log('Device discovered: ', device);
}, error => {
  console.log('Cannot scan.');
}, () => {
  console.log('Scan completed. But this event will never fired.');
});
```

don, 2018-06-05:
I don't really want to change the API by adding another callback or break the contract by send a complete message through the existing callback, since that would break existing code.

There are 3 options:
1) Like you mentioned, run a timer that matches the scan time. This is very straightforward and what I normally do.
https://github.com/don/ionic-ble-examples/blob/master/scan/src/pages/home/home.ts#L35
2) Use [startScan](https://github.com/don/cordova-plugin-ble-central#startscan) and [stopScan](https://github.com/don/cordova-plugin-ble-central#stopscan). I do this when I need more control.
3) Wrap the scan function. Your example had issues, but I'm sure this could be made to work.

I'm not sure about your Ionic vs Cordova questions. Other people have written the Ionic wrappers they work well but I haven't looked Into the implementation.

Domvel, 2018-06-06:
np. this is just a feature request / suggestion. For my case, I wrapped the method. Btw. I wrapped the whole library in my custom BLE service to simplify the usage. For example the device id is managed automatically. So my read / write methods do not need the device id anymore. :) And for the scan, I implemented the completed callback. 👍 All fine. But it's up to you to decide to close this issue.