don/cordova-plugin-ble-central #587: Reset BLE cache

Domvel, 2018-06-05:
Is there a way to clear the BLE / Bluetooth cache of the system? Is there an API?
Android has sometimes a cash problem. Especially if the peripheral changed something in the service / characteristic directory. In my case there occured many characteristics which the device does not really have anymore. Including wrong links / references. Result: Can't access read or write.
In this case you have to go to the settings and reset the bluetooth manually.
But is there a programmally solution?

don, 2018-06-05:
I don't think there's any solution in iOS other than having the user turn Bluetooth on and off.

The nRF Connect app on Android has a "Refresh Service" menu option so there is a way to do it on Android. The code for nRF Connect is open source so it's possible to see how they implement it.

[image]

Domvel, 2018-06-06:
This method of the nRF Connect App works. I tested it right now. Thanks for the tip.

don, 2018-06-06:
@Domvel I though it was open source. Maybe something changed. Anyway maybe nRF Toolbox or their common library has the same function? https://github.com/NordicSemiconductor/Android-nRF-Toolbox  https://github.com/NordicSemiconductor/Android-BLE-Common-Library

Domvel, 2018-06-07:
nRF Toolbox uses nRF connect as plugin. And I found nothing in common library. I have no idea how to refresh services / clear cache. Maybe the android bluetooth API documentation can help? Any other ideas?

Domvel, 2018-06-07:
@don Done! I sent a pull-request.

don, 2018-06-08:
@Domvel how'd you test this?
Which devices are causing you problems?
Are you seeing this for older Android < SDK 23 or all versions?
Do the devices not problematic devices change services without implementing Service Changed 0x2a05 on Generic Attribute Service 0x1801?

Domvel, 2018-06-08:
@don I already tested it in a few tests. It works. It's the same method like in nRF Connect. I corresponded with a developer of this app for support. He also wrote that the spec is clear about cache. For not bonded devices (my case) master should not cache services if a device has service changed characteristic. Android ignores this rule and always chaches.
The service changed characteristic does not help in our case. The peripheral resets completely. After reconnection we have the cache problem. A reconnect will not help. Also other bluetooth apps like nRF Connect confirms the wrong characteristics loaded from cache.
Only if I reset the bluetooth enable state in the system settings of Android or the "refresh" method.
I have a Android 6.0 where the problem is nearly 100% occurring. And a Android 7.0 where sometimes it occurs. Another Android 7.0 had never this problem. Maybe it's a Bluetooth 4.0 issue? I can't say it. Only that this clear cache method works.

Test case:
Via the app we can update a control unit. This peripheral has to states. Being in application or bootloader.
On every change from application to bootloader and vice versa, the controller re-initialize the bluetooth chip with a few other characteristics. Here's a example of an update. The controlle is in application more. And we want to update this application. The app is connected with the peripheral.
1. The app write a command to a characteristic to start the bootloader of the controller. 
2. The app waits a few seconds until the controller is completely started into the bootloader.
3. The app reconnect to the peripheral to get the correct characteristics / handles. Otherwise the notifications will not work. 
Now we have the cache problem on some android devices. Same case after the software update and startup from bootloader to application. Because of changed services. 
In this case, I invoke the clear-cache-method and reconnect. Fixes the problem.
So far to the background story. :)

You could hide this method from the documentation if you want. But I and maybe other need it.
The implementation does not affect other stuff in the plugin. It's a capsuled method.
There is just no guarantee that this works for all. There is no support. In my case it's the only one solution we have. And similar to the nRF Connect method "Refresh services".
It would be great if this is in the upcoming release as soon as possible. 

Note:
The clean-cache-method is a hidden methode of the BluetoothGatt Class of the Android Bluetooth API. Only accessible by reflection method. It's an asynchronous method and it may take some time before it's complete. There is no complete callback. The success callback is called immediately after invoked.
So the usually way to use this method is: 
Call it while connected and wait a few seconds. Then re-connect (disconnect -> connect) from peripheral.

don, 2018-06-08:
@Domvel I'm not going to remove the feature, I'm just trying to duplicate here

Can you try this new branch and see if the JSON returned to ble.refreshDeviceCache contains the new services in the callback?
 
    cordova plugin rm cordova-plugin-ble-central
    cordova plugin add https://github.com/don/cordova-plugin-ble-central#refresh-services

Domvel, 2018-06-08:
@don No problem. I had not thought that either. My intention was to obscure this method like the Android API do. To avoid support requests about this dirty insecure method. 😄 But I see, you improved my documentation about this. great. 👍 Btw. I forgot to add this in the table of contents links above.

Testing the branch ...

Domvel, 2018-06-08:
Cannot test because:
https://jcenter.bintray.com/android/arch/core/common/1.0.0/common-1.0.0.jar

```json
{
  "errors" : [ {
    "status" : 404,
    "message" : "Could not find resource"
  } ]
}
```

domske, 2018-06-08:
Perhaps someone removed the `Android Arch` from `jcenter`. Use the maven google server. 

### Solution
Go to `<project>\platform\android\build.gradle` and move `jcenter()` behind `maven.google.com`.

Like: 
```text
allprojects {
    repositories {
        maven {
            url "https://maven.google.com"
        }
        jcenter()
    }
}
```
build again. It should work now.

Domvel, 2018-06-11:
@don It seems to work. (a few tests) Thanks. Great that the gatt discovers the services directly after the refresh. But I don't know when the refresh method returns the boolean. After invoked or after refresh completed. The developer of nRF Connect means it does not have a complete callback. Maybe the success returns directly after the method invoked. In worst case (big data of services) it can be a race condition. I found nothing in the documentations. Of course, it's a private method.

But anyway ... this is a better implementation (with gatt.discoverServices()) in my opinion. Do not care of possible issues for now. It works. Good job. Can be released. 😃

Domvel, 2018-06-11:
After a few tests I found a weird problem with the returned new json of discovered data.
After connection with the `connect` method I get an object with all correct services / characteristics. 
And after `refreshDeviceCache` the new json characteristics index is sometimes incomplete. 
And now I have a complete json but a notificaton fails with `Set notification failed for <UUID>`

Another weird thing is that the `advertising` of the json is empty on `connect` method. On `refreshDeviceCache` it has the correct object. `{CDVType, data}`

I think I know why is the `refreshDeviceCache` method fails. Because of `gatt.discoverServices()` before the `refresh` method is completed. **NOTE** I already wrote: The developer of Nordic has informed me that this method `refresh` has no complete callback .The return value indicate only the successfully invoke of this method. But the cache clearn is asynchronous and it may take some time before it's complete.
When you'll call `gatt.discoverServices()` before it's done, it will abort and start discovery again. When after it's done, it will result immediately wich cached services. 

I think, the only solution with gatt.discoverServices() is to delay it a few seconds. 
1. Call the `refresh` method.
2. Wait 10 seconds. 
3. Call 'gatt.discoverServices`

don, 2018-06-11:
refresh returns a boolean, but the function it calls doesn't, maybe adding a delay will help https://android.googlesource.com/platform/frameworks/base/+/ddf7e4756c31d0ed90802f98abeaa79df6d16b2a/core/java/android/bluetooth/BluetoothGatt.java#1063

`connect` is returning the peripherals correctly. The advertising data is an ArrayBuffer. The data is there but it logs as {}.  `refreshDeviceCache` needs to wrap the callback to handle nested ArrayBuffers correctly.

don, 2018-06-11:
On branch https://github.com/don/cordova-plugin-ble-central#refresh-services ble.refreshDeviceCache now takes a timeout argument.

Domvel, 2018-06-12:
It works. Thanks don.

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios