don/cordova-plugin-ble-central #588: TAG the current master and update NPM package.

Domvel, 2018-06-05:
@don Please could you tag the current master (because the merge #568) and update the NPM package. Also the ionic wrapper. 😃

don, 2018-06-05:
I'll try and get a release out soon. I haven't messed with the Ionic wrappers. I'd appreciate if you submit  a pull request to their repo with the new function.

Domvel, 2018-06-06:
Me too. I have no experience how the workflow is for the ionic wrapper. But I could try to post a pull-request for the ionic-native/ble wrapper repo. It feels lika a looong way from a change in this repo to ionic. 🤔 I think the ionic wrapper needs the npm version of this plugin. Otherwise it's not consistent. I can't post feature changes without knowing the version. I can't add a new method which does not yet exist in the cordova plugin.

don, 2018-06-07:
Released 1.2.0

closed 2018-06-07 by don