don/cordova-plugin-ble-central #590: Connect to a bluetooth device which connected by other iPad or iPhone problem

Lanseer, 2018-06-07:
the problem happened like that,I have an iPad and connected to a bluetooth device, then i use my another iPad to connect to the same bluetooth device, then i can not get a result returned by the function ,my code is below:
this.ble.connect(device.id).subscribe(success => {
               console.log('connected !')
            },
            failure => {
                console.log('failed to connect !')
            });

don, 2018-06-07:
@Lanseer not all Bluetooth peripherals allow multiple connections. Can you try using [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer/id557428110?mt=8) on both iPads to see if you get different results?

Lanseer, 2018-06-08:
@don Yes,I did.but my point is not about multiple connections, it is about getting  result when connect to a bluetooth device which connected by other iPad or iPhone,I want get console.log success or failure,not waiting for all the time,I hope receive your reply, thanks for your help!

Lanseer, 2018-06-08:
@don maybe i should set a time limit for connection, like: if there is no result after 10 seconds,then i kill the connection request and show a toast 'failed to connect'😄

don, 2018-06-08:
Yes, if you're using cached scan info and try and connect, iOS might take a long time to connect or fail. 

Setting a timeout in your code could solve this. #425 is to add a generic timeout but that's not likely to be implemented anytime soon.

Another option is to make sure your scan results are fresh. Many BLE devices stop advertising when a central device connects. If you do rescan, the 2nd iPad might shouldn't see the peripheral and know not to connect.

closed 2018-06-08 by don