don/cordova-plugin-ble-central #597: Connect automatically (without user action) based on device name fails due async execution of ble.scan

I-Connect, 2018-06-09:
Hi, I want to connect automatically based on a hard coded known device.name without the user needed to press a connect button. 

On device ready run the scan:
`scanForDevices: function(){
	    	ble.scan([], 10, app.onDiscoverDevice, app.onError);
	    	app.getDevIdFromList();
	    },`

Construct an array with all devices discovered:
`onDiscoverDevice: function(device) {   	
	    	deviceList[deviceNr] = [device.name, device.id, device.rssi];
	    	deviceNr = deviceNr + 1 ;
	    },`
And when the ble.scan function finishes run app.getDevIdFromList() which checks if my device name is in the list and if yes start connecting:
`getDevIdFromList: function(){
			for (var i = 0; i < deviceList.length; i++) {
				if (deviceList[i][0]== "myDeviceName"){
					myDeviceDetected = "true";
					myDeviceId = deviceList[i][1];
                                        app.connect();
				}
			}
			if (myDeviceDetected == "false"){
				app.onBtDetectionError();
			}
	},`

The problem seems to be that getDevIdFromList is being called before ble.scan finishes (running async?) resulting in an incomplete or even empty array being processed by getDevIdFromList throwing the onBtDetectionError even when my device is available.

Any idea how to fix this?
thx!
(ps not sure why the code is not indented properly, shows oke in the write pane...)

don, 2018-06-09:
@I-Connect this is a much better question for StackOverflow since it's not a bug. 

Post the question on StackOverflow and link to the question here so I can answer it for you.

I-Connect, 2018-06-09:
fair enough, done: [https://stackoverflow.com/questions/50776972/cordova-ble-connect-automatically-without-user-action-based-on-device-name-fai](url)

allthough it would be nice if there would be a function within  the plugin which can connect based on a device name?  ;-)

don, 2018-06-09:
Answer https://stackoverflow.com/a/50777755/1416463

closed 2018-06-09 by don