don/cordova-plugin-ble-central #601: Write fn can support other data type ?

emjoyLee, 2018-06-12:
ah hello , i'm a novice use that.
Now i have a requirement.I need connect a BlueTooth Printer to print some word and QRCode.
I was connect success and write some Number.
When i print String , it dose not work.
Which code i write can print them. String and QRCode.

Please help me. Thanks 

code
                    `writeBT (data) {
                        function stringToBytes(string) {
                            var array = new Uint8Array(string.length);
                            for (var i = 0, l = string.length; i < l; i++) {
                                array[i] = string.charCodeAt(i);
                            }
                            return array.buffer;
                        }
                        
                        let serve = '18F0'
                        let chars = '2AF1'
                        let str = '12345678987654321'

                        ble.write(this.selectId, serve, chars, stringToBytes( data || str), (res) => {
                            console.log('write success', res)
                        }, (err) => {
                            console.log('write err', err)
                        })
                    },`

env 
MDlogic Printer
BT58s

ios11.4
android7.0
cordova8.0.0

Domvel, 2018-06-12:
@emjoyLee Hi. This is a question for a forum. But in this repository please only report issues like bugs or requests. Not questions about your specific project. Your question has nothing to do with this cordova plugin. You could contact the customer support of the printer or try to ask in a forum. Don't use this issue-tracker as forum. Thanks :)
You code should work. It writes the string '12345678987654321' to the characteristic '2AF1' as array buffer. Assuming your printer supports this. But idk.

don, 2018-06-13:
@emjoyLee write takes an ArrayBuffer since Bluetooth characteristics are just bytes. You need to figure out how to translate the data you need into bytes. If the printer accepts string data you can use `ble. stringToArrayBuffer(str)` but it only works for ascii data. Better code would use something like [TextEncoder](https://developer.mozilla.org/en-US/docs/Web/API/TextEncoder) or [text-encoding](https://github.com/inexorabletash/text-encoding).

Like @Domvel said, the Bluetooth code looks OK. You need to figure out what the data the printer needs to print text and QR codes. I see a lot of questions like this. I don't have any of these printers so I have no idea how they work. Try Stack Overflow to see if you can find someone else using a similar printer. If you find info about this, please let me know so I can point other people to the solution.

closed 2018-06-13 by don