don/cordova-plugin-ble-central #602: Build failure with 1.2.1: lambda expressions not supported in -source 1.6

kellycampbell, 2018-06-12:
Hi Don,

The recent patch release caused this build failure: 
```
:compileReleaseJavaWithJavac/eng-build-pool/jenkins/workspace/xfinity-mgmt-app-android-release-4.0/cordova/platforms/android/src/com/megster/cordova/ble/central/Peripheral.java:199: error: lambda expressions are not supported in -source 1.6
12:49:38                         handler.postDelayed(() -> {
12:49:38                                                ^
12:49:38   (use -source 8 or higher to enable lambda expressions)
12:49:38 1 error
12:49:38  FAILED
```

We're using cordova-android 6.4.0 and android sdk tools v 27

don, 2018-06-12:
@kellycampbell I'm assuming switching your build to use source 8 will be too disruptive? I think there's only one lambda so it should be easy to fix. I'll get new release out soon.

closed 2018-06-13 by don

kellycampbell, 2018-06-13:
We can't switch on our current release cycle without causing all kinds of build and retest pain, so we changed our semver spec to use ~ instead of ^ so we'd stay with 1.1.x instead of 1.2.x of the plugin. 

I figured you'd still want to know for avoiding this issue with other users of the plugin in a dot patch release. Thanks for the quick response!

don, 2018-06-13:
This is fixed. I just pushed version 1.2.2 to npm for you.

commit a4905af2 (2018-06-13): remove lambda for 1.6 source compatibility fixes #602