don/cordova-plugin-ble-central #605: bonding

don, 2018-06-13:
Adding support for bonding has been a feature request. This can be implement on Android. There are no APIs on iOS. Adding some implementation notes so they don't get lost.

`ble.createBond(deviceId, success, failure)`
BluetoothDevice.createBond returns immediately. Need to add an intent filter for BluetoothDevice.ACTION_BOND_STATE_CHANGED and register a broadcast receiver. 

Bond state change notifications could be sent to the success callback or we could make the user register another listener `ble.startBondStateChangeNotifications` which could send the MAC address and the bonding state. It's a cleaner Javascript API if the plugin handles the bond state change notifications and sends them to the success callback of create bond.

Assume the OS will handle the pin negotiation. Revisit when that's not good enough.

`ble.getBondedState(deviceId, success, failure)`
Returns BOND_NONE, BOND_BONDING, BOND_BONDED to the success callback

The peripheral JSON should be modified to include the bonded state.

BluetoothDevice does not have a public unbond method.

imaginae, 2019-07-05:
Hi Don, did you finally worked on createBond? There are problems coming when boards are pin protected... We are trying to override module with a setPin method, but we have not success on using this method to avoid errors on write. Regards

don, 2019-10-18:
@imaginae I have not worked on this. Perhaps you can bond the device outside of the app then use [ble.bondedDevices()](https://github.com/don/cordova-plugin-ble-central#bondeddevices)?

imaginae, 2019-10-21:
Hi Don,

we don't use bond, finally we avoided to use pinage as per errors on
setting/getting them from board... We finally work with attributes inside
boards.
In any case, there some other issues at this point that some people in our
team have solved, do you want me to share it or better if we create a push
in a branch to update?

Thanks!

*Juanjo Raya*
Project manager
www.imaginae.net
------------------------------

Avda Francesc Macià 30 Esc A 1er 1a

Sabadell (Barcelona)
Tlf 93 537 74 44
[image: Facebook]
<http://www.facebook.com/pages/Imaginae-Solutions-InternetEcommerce/141782755888519>
[image:
Twitter] <http://twitter.com/#%21/ImaginaeSol>
Skype: [image: Skype] imaginae

don, 2019-10-21:
That'd be great if you could submit a pull request with your updates.

peitschie, 2023-02-18:
Feature has been implemented in 1.6.3

closed 2023-02-18 by peitschie