don/cordova-plugin-ble-central #606: Write stops working after requestMTU is called

tempestwf, 2018-06-15:
Hi

We are very glad for the release of 1.2.2 but we are having some issues with the new functionality.

We have been needing the requestMTU functionality, however, we find that our write operations start timing out after it's called.

We tested quite some time, all but one of the of the write operations failed, either with triggering the failure handler function after several seconds, or not triggering it at all.

1 write request got through, however, it may have been triggered before MTU was called (due to tweaks we were doing to try to figure out the problem).

We turned off the requestMTU functionality in our app, and everything went back to normal.

Thanks for turning out the new version.

Will

don, 2018-06-15:
I don’t know much about requestMtu. The Android docs say it’s to stop write without response from truncating data which makes sense. Unfortunately the code isn’t checking to see if this succeeds. I’d guess that setting too large an mtu might upset some peripherals?

FWIW, I’ve been able to write 500 bytes of data to some peripherals without changing the mtu. Android transparently breaks the data up into multiple blob writes. 

If you figure out how to get requestMtu to behave better LMK and we can update the docs.

tempestwf, 2018-06-17:
Hi Don

That is a good tip. I will try it with different sizes. Right now I am passing a value of: 512

I have tried that same value with cordova-plugin-bluetoothle plugin and it worked there. The MTU settings are needed by us because the device we are working with sends too large of messages (as notifications), which then get broken up, and get returned in the wrong order when on Android (leading us gluing them back together wrong). I believe I mentioned this in another issue thread here, but just bring it up again for clarity.

We thought we could solve it with cordova-plugin-bluetoothle however it has it's own bugs which are preventing it from working when packaged into an APK.

It's always something.

Hope you are having a great weekend.

Will

Domvel, 2018-06-19:
@tempestwf You can't write anymore after requestMTU? In my case I have to set the mtu size greater that one message (+ ble meta byte) when writing to a characteristic with more than 20 byte. And btw. another benefit is to speed up the transmission. 
I set the MTU Size always to 512 byte. And we have no issues with it. (yet) The most characteristics have 20 byte and only needs the minimum mtu size of 23 byte (default). Only our update characteristic has about 200 byte. I send the update (500 kb file) in chunks and the peripheral controller software handles it. Works great. It takes about 15 minutes. 😄 I have to improve this in the future by e.g. zipping and multiple streams. Whatever ... this is my story. 

You wrote that the bluetoothle-plugin is working in your case with the same setup? (mtu: 512) We have to compare the write methods. Which data are you sending?

tempestwf, 2018-08-20:
Hi Domvel

Sorry for my late reply. For the moment we are using the Rand Dusing plugin
instead as we fixed the bugs we were having with that one before we managed
to fix the bugs we were having with this one.

If we switch back in the future this will be good information to have.

Thanks very much!

Will

GiovanniVisentini, 2018-11-13:
Hello,
I had the same problem, and I solve it with a delay.
the success callback is called when the request is sent:
https://github.com/don/cordova-plugin-ble-central/blob/ac55995dc9e9d19f231c13e6bbfb9ba24b076480/src/android/BLECentralPlugin.java#L440

But it should be called when the android system tell you the request result: 
here: https://github.com/don/cordova-plugin-ble-central/blob/ac55995dc9e9d19f231c13e6bbfb9ba24b076480/src/android/Peripheral.java#L167

You should also consider to return a Promise of int since the device could negotiate a different mtu length.
Example: you request 512, the maximum supported by the peripheral is 200, android will change the mtu to 200 and pass the value into the on onMtuChanged..

Best regards
Giovanni

tempestwf, 2018-11-17:
HI  Giovanni

That is very handy! If we use this plugin again in the future we will use
that fix.

Thanks for the information.

Will

agren, 2019-11-06:
I had the same problem so I modified the plugin to call the success callback when onMtuChanged is called (As suggested by others). The actual MTU size is passed to the success callback.

This should eliminate the need for artificial delays after calling requestMTU. At least it did for me.

I've made a pull request (#715 )

closed 2022-01-24 by peitschie