don/cordova-plugin-ble-central #607: With version 1.2.2 I need to enable location services on my tablet

tempestwf, 2018-06-15:
Hi

Thanks very much again for turning out the new version. We have found that with this version plugin will not work, and throws an error if the location services are not enabled.

This error was thrown to the error handler of the enable method.

I can't say for certain if this issue existed in a previous version or not, as we added the enable call only after upgrading to 1.2.2.

I am testing on a Samsung Galaxy Tablets running Android version 7.0.

All the best

Will

don, 2018-06-15:
1.2.1 added the location services check. Prior to that, people would not get scan results when location services were disabled. 

Can you try version 1.2.0 on that tablet and see if you get any scan results with location services disabled?

I probably need to move that check to scan related functions.

tempestwf, 2018-06-17:
Hi Don

Thanks for the write back. I will check this out and get back to you.

All the best

Will

doug-a-brunner, 2018-06-21:
@don, I ran into this as well. I can confirm the problem appears with commit [577bd15](https://github.com/don/cordova-plugin-ble-central/pull/595/commits/577bd15f4fc5ede186a65662c526efc769daae0b#diff-16acba3b47306900a48cec22c99a92a3). Interestingly, on multiple Kindle Fire 5th generation units we have here, scanning for BLE devices works just fine with location services off (the default) - this change broke our ability to operate with the default setting.

don, 2018-06-22:
@doug-a-brunner I’ll look at this late next week when I’m back from vacation. 

Do the Fire devices request permission like Android 6.0+ or are all the permissions in the manifest? Is there another way to tell if it’s a Kindle or Android?

airbly, 2018-07-09:
Hi,

On an Android 8 based Samsung Galaxy phone, I am experiencing the following:

v1.2.0 - Scan works great with location services disabled or enabled
v1.2.1 - Scan fails (calls the fail callback) with location services disabled.  Works fine when location services are enabled
v1.2.2 - Scan fails (calls the fail callback) with location services disabled.  Works fine when location services are enabled

The app does not prompt me to enable location services in either version.

I hope that helps.

- Chris

AvidanBaruchiN, 2018-09-02:
I encountered the same problem.
Location permission is granted. but when location is off, not working and not prompting.

don, 2019-10-18:
Fixed in #633

closed 2019-10-18 by don