don/cordova-plugin-ble-central #608: Android "disconnect()" not works properly

LabDeve, 2018-06-19:
When I run the method 'disconnect' it goes right, but if i check my ble device, it isn't disconnected...

Logcat:
```
06-19 14:58:22.876 27440-27501/it.user.myproject W/BluetoothGatt: Unhandled exception in callback
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattCallback.onConnectionStateChange(android.bluetooth.BluetoothGatt, int, int)' on a null object reference
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:229)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:573)
```

Plugin: `"cordova-plugin-ble-central": "^1.2.2"`

Android device: `Samsing SM-G930F (Android 7.0, API 24)`
Ionic info:
```
cli packages: (/usr/local/lib/node_modules)

    @ionic/cli-utils  : 1.19.2
    ionic (Ionic CLI) : 3.20.0

global packages:

    cordova (Cordova CLI) : 8.0.0 

local packages:

    @ionic/app-scripts : 3.1.9
    Cordova Platforms  : android 6.4.0 ios 4.5.4
    Ionic Framework    : ionic-angular 3.9.2

System:

    Android SDK Tools : 26.1.1
    ios-deploy        : 1.9.2 
    ios-sim           : 5.0.13 
    Node              : v9.11.2
    npm               : 5.6.0 
    OS                : macOS High Sierra
    Xcode             : Xcode 9.4 Build version 9F1027a 

Environment Variables:

    ANDROID_HOME : /Users/user/Library/Android/sdk

Misc:

    backend : legacy
```

Domvel, 2018-06-20:
#473

don, 2018-06-29:
@LabDeve I'm closing this since it's a duplicate of #473. I think that issue lists a good solution, I just need to implement it.

closed 2018-06-29 by don