don/cordova-plugin-ble-central #609: ble.scan/startScan intermittently failing to pick up devices

leydar, 2018-06-20:
**Expected Behaviour**
When scanning with:
```      
      this.ble.scan([], 30).subscribe(device=>{
        console.log("scan", device);
      });              
```
The device should be picking up multiple BLE devices which are continuously advertising. It's currently picking up 13 for example. 

**Failure Mode**
Sometimes this just isn't the case and the console log function is never called. At the same time I have a BLE scanner showing the devices advertising. I have also had apps running on two different phones (iOS 11.4, Android 6.0) whereby one will detect devices and the other won't. This is not constant behaviour.

**Steps to Reproduce**
Create a new Ionic app, include cordova-plugin-ble-central and scan for BLE devices. The failure is intermittent but frequent.

**Context (Environment)**
cordova-plugin-ble-central 1.2.2 "BLE"

Huawei Nova Young 4G LTE MYA-L11
Android: 6.0 Marshmallow

iPhone SE
iOS 11.4

ionic/cli-utils  : 1.19.2
ionic (Ionic CLI) : 3.20.0
cordova (Cordova CLI) : 8.0.0 
ionic/app-scripts : 3.1.8
Cordova Platforms  : android 
Ionic Framework    : ionic-angular 
Android SDK Tools : 
Node              : v9.11.1
npm               : 6.1.0 
OS                : Linux 4.15

radoralai, 2018-06-20:
Hi leydar,

Is the failure appear essentially on iOS?
If so, I've already been facing to it.
The issue is a random peripheral detection.
This is what I did to solve the issue:

In the BLECentralPlugin.m

```
- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {

    [peripheral setDelegate:self]; // I added this line to fire the issue 
    [peripherals addObject:peripheral];
    [peripheral setAdvertisementData:advertisementData RSSI:RSSI];
```
@don How do you think about that?

leydar, 2018-06-20:
It's actually happening on both devices. But currently not on the Android. I'm thinking it might be something in the cache from a previous bad connection/disconnection that's preventing the scan working. 

But could you explain in more detail what you mean by a random peripheral detection? The issue I'm having is that it isn't detecting anything at all.

radoralai, 2018-06-20:
I'm working on an iOS app that should detect a BLE device.
The Android version detects it everytime but not iOS which detects one or 2 times of 10 scans.
On failure, I got trace like below:

```

> Discovered {
>     advertising =     {
>         kCBAdvDataIsConnectable = 0;
>         kCBAdvDataManufacturerData =         {
>             CDVType = ArrayBuffer;
>             data = "BgABCSAASB03ZtFGqGYb1AEcYH1YRcj+zbE=";
>         };
>     };
>     id = "8A5CA410-1757-DDF5-2F4C-4F23477638DE";
>     rssi = "-89";
> 

```
Neither 2 scans between short period nor a long long time interval was the cause.
The weird thing was that the application GATTBrowser detected permanently the device

leydar, 2018-06-22:
Yeah, in the cases where the BLE plugin didn't pick up any signals other scanning apps on the phone did. Have you looked at alternatives?

radoralai, 2018-06-22:
Yes. I just set up the peripheral delegate as shown above

Télécharger Outlook pour Android

leydar, 2018-06-27:
I have a mobile currently in a fail state. It was connected to a BLE device which dropped the connection. I have reinstalled app on the mobile but the BLE scan still reveals nothing. Running a scanner on the same mobile shows no nearby devices. Although running on a different phone shows a list of them. So whatever has happened has locked the adapter. 

I know resetting the bluetooth adapter on the phone will likely clear this state. I'd rather use this opportunity to find a software solution or root cause. I'm open to suggestions.

don, 2018-06-29:
If you're connected to a device it won't show up in the scan results. iOS and Android have difference caching problems you might run into. Also, depending on the scan value and the advertising timing you might not see a device. This plugin delegates all scanning to the underlying OS. Sometimes rebooting the phone fixes random problems, usually on Android. However some users were seeing odd iOS behavior on some devices, see #575.

@leydar ideally you can write time simplest possible Cordova app to duplicate the problem. If I can duplicate it, I'll have a better chance of fixing it. Unfortunately this issue is probably device specific.

@radoralai I'll check out your code change to see if this explains what's happening.

leydar, 2018-07-03:
@don I'm pretty sure it was due to me trying to direct connect to a previously found device. Having removed that code and reverted back to scanning for a device each time it seems fine. That's not definitely what the issue was but so far it looks to be working ok. I'll stick with it anyway and let you know if I find a specific cause.

don, 2018-07-03:
@leydar thanks for the update, I'm going to close this issue.

With recent versions of the plugin, you can pass a MAC address to ble.connect and ble.autoConnect with Android, but iOS needs to know about the device. 

For iOS Scanning again is an easiest solution. If the device is already connected to the phone, you can get it using [ble.connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices) or [ble.peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers) on iOS.

closed 2018-07-03 by don