don/cordova-plugin-ble-central #610: Understanding cycling speed and cadence sensor data

jabelcs, 2018-06-25:
Hello all.

First of all I would like to thank @don for this awesome plugin.

Second I need to say this is not an issue but mainly an help request and that for me the plugin seems to be working fine.

I'm starting to use bluettoth technology for the first time so I'm very newbie.

I'm trying to build an app with Cordova that mesasures speed and cadence from an indoor cycling sensor.

I'm writing the code using Visual Studio 2017 and a Samsung Android tablet to test. From the plugin I can connect to the bluetooth sensor and retrieve its data.

I retrieve the following properties and characteristics from the sensor:

name (string) : Dual_SCA06232 (string)
id (string) : D0:EB:FD:5A:48:5A (string)
services (string) : object
    0 (string) : 1800 (string)
    1 (string) : 1801 (string)
    2 (string) : 1816 (string)
    3 (string) : 180f (string)
    4 (string) : 180a (string)
characteristics (string) : object
    [...]
    3 (string) : object
        service (string) : 1801 (string)
        characteristic (string) : 2a05 (string)
        properties (string) : object
            0 (string) : Indicate (string)
        descriptors (string) : object
            0 (string) : object
                uuid (string) : 2902 (string)
    4 (string) : object
        service (string) : 1816 (string)
        characteristic (string) : 2a5b (string)
        properties (string) : object
            0 (string) : Notify (string)
        descriptors (string) : object
            0 (string) : object
                uuid (string) : 2902 (string)
    5 (string) : object
        service (string) : 1816 (string)
        characteristic (string) : 2a5c (string)
        properties (string) : object
            0 (string) : Read (string)
[...]

Reading the [Bluetooth website](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.csc_measurement.xml) I understand that the values I need are returned by service uuid 1816 and its characteristic uuid 2a5b.

After connecting to the sensor I retrieve the values using the following code:

```
ble.startNotification(device_id, '1816', '2a5b', function (buffer) {
    var data = new Uint8Array(buffer);
    console.log(data);
}, onError);
```
So from the console I get the following values:

"-3,70,3,0,0,55,255,70,3,94,65"
"-3,73,3,0,0,23,0,73,3,112,69"
"-3,76,3,0,0,250,0,76,3,144,73"
"-3,79,3,0,0,224,1,79,3,190,77"
"-3,82,3,0,0,201,2,82,3,251,81"
[...]

Which off this value do I need to use so that I can calculate speed, cadence and distance from the sensor?

Thanks in advance.

amcolash, 2018-11-08:
I am working on the same issue and wanted to share what I have learned so far.

You have your data, you just need to pick apart those values from the buffer now. Since each value listed above is an unsigned 8 bit number you need to combine some together. The first 8bit value (as listed on the spec you mentioned) is the flag info. an 3 == 11 (in binary), so you have both wheel revs + crank revs present. Next take the next 32bit int, or 4 values and merge for cumulative wheel revolutions. Convert all to binary then add, then convert back.

70 = 0100 0110
03 = 0000 0011
00 = 0000 0000
00 = 0000 0000

therefore our number is 0100 0110 0000 0011 0000 0000 0000 0000 which converted = 10600600000.

I think I got that one right, but that is the rough idea at least

Here is a little list in a more readable fashion:
Flags: 8bit   (value 1)
Cumulative Wheel Revolutions: unit32  (value 2-5)
Last Wheel Event Time: unit16  (value 6-7)
Cumulative Crank Revolutions: uint16  (value 8-9)
Last Crank Event Time: uint16  (value 10-11)

Now to get things like speed / distance / etc, that is a whole different can of worms... I have been taking a look into this codebase where someone did exactly that https://github.com/kloppen/trainerpi

Cheers!

amcolash, 2018-11-12:
More updates, check out my project bikevr and how I grabbed the data: https://github.com/amcolash/BikeVR/blob/master/js/bluetooth.js

a-milburn, 2019-06-13:
@amcolash I've been struggling with this task for a couple of days. Your posts so far have given me a lot of insight...but i'm seeing some strange results with the values. Do you have an updated link (the one you posted above gives a 404). Reading the cadence data is much trickier than i thought.

amcolash, 2019-06-14:
Funnily enough I have been working on a different project lately but the same Bluetooth device. Not sure why that link is dead as the repo is still there... Anyways, this is a simpler example: https://github.com/amcolash/oled-bike-computer

a-milburn, 2019-06-19:
Thanks @amcolash I appreciate the help.

closed 2019-10-18 by don