don/cordova-plugin-ble-central #613: Can I get Bluetooth 4.0 Hm-10's address from ios?

kotran88, 2018-06-27:
On android, I use to get Hm-10's address well.
address is like 12:34:56:78:9A:BC

but the problem is that...
the id value that I can get from ios is not like above. 
id value is like 03A23620-E31F-FA2C-5E8C-F1C3C35A0A50. 

the value I get from android is like 
{"name":"test888","id":"0C:B2:B7:79:B5:AF","advertising":{},"rssi":-53}
bud the value I get from ios is like
{"id":"9E9EC7E1-6232-7C12-5A12-DC534A24EEDA","rssi":-69,"advertising":{"kCBAdvDataIsConnectable":true,"kCBAdvDataLocalName":"test888","kCBAdvDataServiceUUIDs":["FFE0"],"kCBAdvDataServiceData":{"B000":{}},"kCBAdvDataTxPowerLevel":0,"kCBAdvDataManufacturerData":{}},"name":"test888"}

above values are from one HM-10 device

i'm using same code with hybrid app. 

Did I do something wrong or it is what meant to be. 

because I need to do AT command with arduino after getting information of hm-10 's address. 

I only can do that with android...because I can't get appropriate address from ios

don, 2018-06-29:
You can't get the MAC address with iOS. iOS hides the MAC address and generates a UUID. The generated UUID for a HM-10 will be different on every iOS device. 

If you scan for the Service UUID and connect to the HM-10 you should be able to write your AT commands to the characteristic without a MAC address.

closed 2019-10-18 by don

chaymankala, 2019-10-22:
Hi @don 
I also have a similar issue, I need to connect to my BLE HM11 device using iOS application. I know only the MAC ID of the device. How can I possibly identify my device in the scanresponse?
TIA