don/cordova-plugin-ble-central #614: BLE.autoConnect is not installed

Paul811, 2018-06-27:
I'm currently coding an Ionic app. 
I am on an Iphone 7, and I test the app through Ionic DevApp.

I'm trying to test the autoConnect function in order to be connected each time my RPI is near the phone in order to connect without scanning.
However, when the function is called, it is said that the BLE.autoConnect is not installed. 
I did : ionic cordova plugin add cordova-plugin-ble-central 
and it is said that the plugin is updated. 

Can you help me ?

Domvel, 2018-06-28:
Because this plugin is for cordova. Ionic just wraps it to be conform with angular and typescript.
see https://github.com/ionic-team/ionic-native 

I already sent a pull-request to the ionic-native project and added the `autoConnect` method. It is also merged. But the ionic team has not yet released it. 
Until they do you can workaround this. You can access this plugin directly from the `window` object (global scope) or write your own typed file to extend the plugin.

For example use it like:
```ts
(<any>window).ble.autoConnect(device_id, data => {
  console.log('Connected', data);
}, error => {
  console.log('Cannot connect / disconnected', error);
});
```
It's a bit dirty but it works. And in the future you can replace it with the official ionic style. If they finally release it.
Note you access directly the javascript function. There is no Observable or promise. You have to pass two  functions.

don, 2018-06-29:
For autoConnect to work properly, you need to use the callbacks, even in Ionic. When a device is autoconnecting and disconnecting the success and failure functions are called multiple times.  Unfortunately with an Observable disconnect can only be called 1x, the the Observable is dead and you can't reconnect.

See https://github.com/ionic-team/ionic-native/issues/2573

Paul811, 2018-07-02:
Thank you for the answers ! 

Is it possible to have an example ?
Because I have an error...

Domvel, 2018-07-02:
As wrote. Instead of: 

```ts
// In a class method, ble defined in contructor.
this.ble.autoConnect(deviceId).subscribe(data => {
  console.log('Connected Data: ', JSON.stringify(data));
}, error => {
  console.log('Cannot connect or peripheral disconnected.', JSON.stringify(error));
});
```

do

```ts
// Same place.
(<any>window).ble.autoConnect(deviceId, data => {
  console.log('Connected Data: ', JSON.stringify(data));
}, error => {
  console.log('Cannot connect or peripheral disconnected.', JSON.stringify(error));
});
```

This is one working solution. You directly access the corodva plugin by the window object like in javascript. In typescript you have to cast to any first, otherwise ble is not found. You also could declare this type. But this is a easy and fast solution. A bit ugly / dirty. But this is a temporary solution anyway. Until the ionic-native wrapper is released.

Paul811, 2018-07-02:
I have the following error : "TypeError: undefined is not a functiob (near'...window.ble.autoConnect...').
Do I have to initialize the varialbe "window" ?

Domvel, 2018-07-02:
Maybe you have not the latest version of this plugin? OR you should reinstall the android platform ("platform" folder). Perhaps it's cached a old version without `autoConnect`.

Paul811, 2018-07-02:
1.1.2

Paul811, 2018-07-02:
But I am on an Iphone, so I do'nt think that the problem comes from Android

Domvel, 2018-07-02:
Android is just an example. You should execute `cordova clean` or reinstall the `ios` platform.

Paul811, 2018-07-02:
Yes, but I have the following error when I do `cordova clean` : 
[image]

Paul811, 2018-07-02:
It resolves the problem but I have the same problem in CordovaLib

Paul811, 2018-07-02:
And if I try cordova clean without the android platform, I also have an error

Paul811, 2018-07-02:
The `cordova clean` works well, but your code doesn't... Same error after the clean

Domvel, 2018-07-02:
I just tested it. It works for me. 

```ts
(<any>window).ble.autoConnect(deviceId, device => {
  console.log('Connected', device);
}, error => {
  console.log('Disconnected', error);
});
```

If you get a `ble.autoConnect is undefined` message, you probably have not the latest version of this plugin. Note: Clean the platform folder (e.g. remove it) when you update the plugin. 

package.json
```json
"cordova-plugin-ble-central": "^1.2.2"
```

Paul811, 2018-07-02:
No, I have the following error : `TypeError: undefined is not a function`
I think the problem comes from the `(<any>window)`
Why it doesn't works with this.ble.autoConnect ?

Domvel, 2018-07-02:
Because - as wrote in my first post - Ionic Native has not yet released the wrapper with this methods. They already merged my pull-request with these methods. But not yet released. With `(<any>window)` you access this BLE Plugin directly like a native javascript app with cordova will do.
[Here](https://github.com/ionic-team/ionic-native/blob/master/src/%40ionic-native/plugins/ble/index.ts#L311) are my changes to ionic-native.

Paul811, 2018-07-02:
So I don't hoave to `import { BLE } from '@ionic-native/ble';` ? How do I have to import the plugin ?

Domvel, 2018-07-02:
That's not the problem. You can import the BLE and use the existing methods from this wrapper. It's not a problem to use both. The ionic-native module AND `window.ble` access.

Does the "normal" `connect` method work for you? Not the `autoConnect` method. Please try it to make sure that your usuage is correct.

Paul811, 2018-07-02:
Yes, the normal method works properly. 
I do a scan in a page, when I click on the device and push another page and I send the device to the page, and I call the autoConnect function

Paul811, 2018-07-02:
Should it works on an Iphone through Ionic DevApp ?

Paul811, 2018-07-02:
Why the autoConnect function is referenced in the Ionic Native Doc if we can't use it ? https://ionicframework.com/docs/native/ble/

don, 2018-07-02:
> Should it works on an Iphone through Ionic DevApp ?
Not yet. ble.autoConnect is a new function. The Ionic DevApp is likely using an older version of the plugin.

> Why the autoConnect function is referenced in the Ionic Native Doc if we can't use it? 
The documents are generated from the source code. I'm not sure if that version of the wrapper has been released or not. The 2 project are a bit out of sync.

The function exists in the Cordova plugin, just not in the wrapper so you can use it, just not through the Ionic wrapper yet. See @Domvel's solution https://github.com/don/cordova-plugin-ble-central/issues/614#issuecomment-400973582

Paul811, 2018-07-03:
This is what I did. But it doesn't works with (`<any>window).ble.autoConnect` but it works with `(<any>window).ble.connect`

I searched everywhere but all the plugin are up to date. I deleted the plateform folder and I reinstalled it, but it doesn't works after this.

After some hours of research, I found this : Ionic DevApp currently supports the following plugins:
[...]
cordova-plugin-ble-central 1.1.4 "BLE"
[...]

Could it be the version of the plugin supported by IonicDevApp or not ?

Paul811, 2018-07-09:
I have no error when I deploy the app on an Android device.
In my code, I scan the devices and when I click on a device, I call another page (DevicePage) and I send the device to this page. In the DevicePage, I call the autoConnect function and when the disconnect callback is called, I don't disconnect the ble device in order to be automatically reconnect when the device is in range. Each time I'm connected to a device, I created a page which is visible in the menu.

But the problem is the following : when the device is in range, the app doesn't connect automatically to the device : I have to do the same steps like the first connection.

How do I have to call the autoConnect and where in order to have an automatic re-connection ?

don, 2018-07-11:
You should not need to call autoConnect after disconnect is called. Cordova is going to reconnect automatically and notify you when it happens. For this to work, you will need the new function signature from https://github.com/ionic-team/ionic-native/issues/2573 or  call Cordova directly with (<any>window).ble.autoConnect(onConnected, onDisconnected).

The autoConnect function can live for a long time and may be better in a service instead of on a page.

Paul811, 2018-07-12:
Yes, I use this : 
```
(<any>window).ble.autoConnect(deviceId, device => {
  console.log('Connected', device);
}, error => {
  console.log('Disconnected', error);
});
```

I put this in the constructor of my page, but when I disconnect, I pop the page in order to go to the page of the scan.
Even if the page is pop, the autoConnect function will work ? I want to be automatically reconnected even if I'm not on the DevicePage, and when I'm automatically reconnected, I want to push the DevicePage. How can I do this ?

Is it obliged to use service or it is just better ?

Paul811, 2018-07-13:
Thank you, it works well ! 

But I have to be in a "paring" mode. It means that I have to run the command `sudo hciconfig hci0 leadv 0` and this, each time someone connected to the device. Do you know how can I active this mode forever ? Or how can the autoConnect function can work withtout this mode ? 

Thank you

closed 2019-10-18 by don