don/cordova-plugin-ble-central #615: Disconnect detection is late. Set timeout time.

Domvel, 2018-06-28:
Currently a peripheral disconnect is detected after 20 seconds. When the peripheral disconnects for example by removed the power supply. It's a bit too long.

Is there are way to set the timeout time? Maybe the connect method could take a timeout time as argument. Or is this a fixed time of the android BLE API?

Maybe related to #425.

don, 2018-06-29:
Unfortunately that 20 second delay is Android. I haven't seen any way to make it faster.

Domvel, 2018-07-02:
I have read a few articles about that. The 20 seconds seems to be hardcoded in the Android source code. There are tickets about it [here](https://issuetracker.google.com/issues/37074926) and [here](https://issuetracker.google.com/issues/37119344). That's weak. I don't know if reflection will help.

closed 2019-10-18 by don