don/cordova-plugin-ble-central #616: Original error code

miqmago, 2018-07-01:
Is there a way to get the original error code from device?

Right now I'm using bleno to advertise a characteristic and I would like to decide what to do on the mobile depending on the error.

Right now the plugin is only returning an string: when in bleno I send `ATT_ERR_READ_NOT_PERMITTED` (0x02) in cordova plugin ios I'm getting an string error: `Reading is not permitted` but no error code... Also digging insde the code https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m#L616 I've seen that this is localized error so it means that the string could be different on each country?

Also I'm worried about if there will be different strings in android or ios...

don, 2018-07-03:
@miqmago right now you just get the localized error message on iOS. 

For now, to avoid this, you could use the JSON description of the peripheral check the characteristic permissions before reading or writing. The JSON is passed to the success callback when you connect.

miqmago, 2018-07-03:
@don thanks, for the moment I've made a workaround (very ugly one) that I'm not reading anything from characteristic, I'm always writing data in format

```json
{
    "op": 0,
    "data": {},
}
```

So when I'm connecting to device I'm subscribing to notifications.
I've made an operation which is named `OP_READ`. That makes the device to raise a notification with the needed value.

So I ended always writing data to bluetooth device...

closed 2019-10-18 by don