don/cordova-plugin-ble-central #621: scan() filtered by services seems not to work?

kazbeel, 2018-07-16:
Hi there,

I am developing a simple app to connect to a BLE device. My first approach worked like a charm:

- Scan all (without any "services" filter) BLE devices and list them.
- Select one device and show the peripheral data.

The code except would be like this

```
this.ble.scan([], 5)
    .subscribe(
        device => { devices.push(device); console.log('New device discovered: ' + device); },
        error => { console.log('Error scanning: ' + error); });
```

Here the problem comes when I add a service filter in order to list only certain devices. This service UUID is "0003cdd0-0000-1000-8000-00805f9b0131" so, as API explains, I was supposed to do the following

```
var myService = '0003cdd0-0000-1000-8000-00805f9b0131';

this.ble.scan([myService], 5)
    .subscribe(
        device => { devices.push(device); console.log('New device discovered: ' + device); },
        error => { console.log('Error scanning: ' + error); });
```

It simply does not discover any device, even when I have 2 of them in my desktop.

**What am I doing wrong? Any help would be very appreciated.**

Notes: I have already checked the communication (read and write) with such service+characteristic in both devices.

don, 2018-07-16:
Which OS and phone?

kazbeel, 2018-07-17:
Sorry for not specifying.
- Samsung Galaxy S5
- Android M (6.0.1)
- SDK 23

don, 2018-07-27:
@kazbeel some Android phones have trouble with services filters, that might be the problem.

Use an app like nRF Connect and verify that 0003cdd0-0000-1000-8000-00805f9b0131 is being advertised. The device could have the service but it not be advertising the UUID.

Jingwen-TL, 2019-02-15:
I'm running into the same issue. The device I'm using is 

- iPhone 7s, IOS 12.1.2
- iMac, macOS Mojave
- Cordova 8.1.2

I'm able to scan and connect to the service without filter. I'm also able to find the service UUID through nRF Connect.

Any help would be appreciated.

mazedesign, 2019-06-11:
Same issue:
- iPhone 5s
- Mac mini, Mojave
- Cordova 8.1.2
- XCode 10.1
also we can't get advertised names on iOS, so without that and scanning for services we are completely unable to get a specific device.
Thanks

closed 2019-10-18 by don

DimitrisRK, 2019-11-21:
This issue still persists. Are you going to do anything about it?

ragcsalo, 2019-11-27:
I also can't filter devices by services.... using Android 9 on a Huawei Mate 20. All other functions work great, except this. Any help would be appreciated! :-)

alejoorduz, 2021-04-30:
For anyone having trouble connecting to specific device on ios (also work on android), I solve this issue like this:
```
    this.ble.scan([], 5).subscribe(
      device => this.onDeviceDiscovered(device), 
      error => this.scanError(error)
    );
```

On the function "onDeviceDiscovered" just take device.name and compare it with your device name, if it find it just save the device.id on a global variable, that way you search device by name and save the uuid you need, doesnt matter if iphone change uuid. 

```
onDeviceDiscovered(device){
    if(device.name === 'My_Device_Name'){
      this.iosuuid = device.id;
    }
 }
```
now you can use 

`this.ble.connect(this.iosuuid, device => {.....}`

Hope it helps someone in the future.

obryl, 2021-06-25:
Having the same issue on Galaxy tab A with Android 10. Still can't understand why this issue was closed. The problem is not solved and it makes searching for the device so painfull!

peitschie, 2022-04-11:
For anyone still encountering problems with this, please raise a new issue. Given the age of the bug, it's not entirely clear to me any longer which problem each individual here is experiencing 🙂