don/cordova-plugin-ble-central #624: Trouble connecting when Device has Bluetooth 5.0

jsumeet, 2018-07-20:
I am having trouble when I am trying to make the connection with a phone that has bluetooth 5.0. If I run my app on a phone with bluetooth 4.x, it scans the device that I am trying to connect to and connects perfectly fine, but when I run my app on a phone with Bluetooth 5.0, it can scan the device but it always have trouble connecting to it. It gives me this error:

```
{
     "errorDescription":"Unknown error.",
     "id":"xxx-x-x-x",
     "name":"ABC123,
     "errorMessage":"Peripheral Disconnected"
}

```

I am trying to figure out if this is the problem with the chip in the device that I am trying to connect to or this plugin somehow doesn't support bluetooth 5.0?

don, 2018-07-27:
@jsumeet which device and OS has this problem? It might be a Bluetooth 5.0 problem, or it could be something else. Have you seen the problem on multiple phones?

jsumeet, 2018-07-30:
@don, i have seen it on multiple devices. I tested on iPhone X, Google Pixel 2 XL and Samsung S8

don, 2018-08-06:
@jsumeet I'm hoping to have more time in a few weeks to work on this plugin again. I'll make sure I'm using Bluetooth 5.0 devices for testing where possible. In the mean time, if you have additional details add comments here.

becauseinterwebs, 2018-12-22:
I am experiencing the same issue and getting the same error message.  Happens on 2 different  Pixel 2 phones, but Samsung 7 and 9 and iPhone X are fine. Any progress?

becauseinterwebs, 2019-01-03:
Have been doing some searching and found this list of known issues that may help:

https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues

don, 2019-10-18:
I don't have a fix. Closing old issue. Create a new issue with more details if you're still seeing this problem. Hopefully newer firmware and os updates have fixed this.

closed 2019-10-18 by don