don/cordova-plugin-ble-central #627: Custom function for connecting with PIN

sonnenhohl, 2018-08-07:
Being able to pass a callback function to the connect function where the callback fuction returns a PIN to the connect function when requested.

Domvel, 2018-08-22:
Is this a user authentication issue? Do you want a token (PIN) on connect from the connection callback? Or the user have to send a PIN to connect? Whatever this sound like a feature of your application and not for this plugin. Or do you mean this is a BLE feature? This would be new to me. Could you more describe this issue with examples?

I think you should connect and transfer the PIN over a characteristic. If this PIN is not correct your controller could close the connection or don't answer on other requests to characteristics until a correct PIN.

sonnenhohl, 2018-08-22:
The problem i have today and i believe other users have is that you have to use the default operating system pin input without any control over it. In some versions of android/manufacturer this window don't have a standard behaviour, it appears over the screen or sometimes just appears in the notifications bar.

My idea is to be able to pass a pin_function() to the connect() function where when the device detects it needs a pin the pin_function() is called where you can create a custom screen for pin input and after the input this pin_function() return the pin to be used during the connect.

This feature will allow you to create custom screens for pin input and also direct connection with fixed pins.

Domvel, 2018-08-22:
Ah, now I understand. The Bluetooth PIN for devices on bonding. I never faced this, because I use this BLE plugin to connect directly to a device (without bonding). No PIN is required in my case. 
But you're right. I know there is a PIN feature for bluetooth devices. I don't know if this is a BLE feature or a level under BLE (Bluetooth). Maybe this is only handled by the operating system. 
Sorry, I'm out. 😄  I can't help you. I've no experience with this. 

But wait ... you get a PIN prompt dialog of the OS when you invoke the ble.connect method, right?
And you want to replace the native PIN dialog of the OS  (like JS preventDefault) with a custom dialog. 

Let me think aloud:

- You don't know the PIN before connect request. And every connect request generates a new PIN on the other device. This means we need a method with only reqeust a connection. Wait for the PIN on the other device and call the connect method with the current active PIN. This is only required for the first bond.

- I'm not sure if we can handle the PIN dialog. Maybe there is no Android API to preventDefault or return that the dialog is already handled. 
 
There is a method to get the bounded devices. I think, in this case you should bond your device in the OS settings and in the app you can access the bonded devices. 
But as wrote, I've not enough experience with this feature. Sorry. :)
You could search the internet, wait until this feature is implemented here, deal with the current behavior or find a workaround.

sonnenhohl, 2018-08-22:
In the company i work we create eletronic products where the PIN can be defined by the user inside the product throught the IHM or another software with usb connection.

When i connect to the device, before bonding the operating system detects a PIN is required because im unable to write to a specific characteristic, on this moment a PIN is requested to me and after that if the PIN is correct the device get bonded with phone.

On future connections i just inform the device ble address and the operating system knows it is bonded and connects directlly with the encryption keys already generated during the first connection.

My only issue really is that i would like to customize this PIN screen because the differences of behaviour between android versions (notification bar, screen ...).

I dont know the apis of android and ios if they have a function to pass the PIN hardcoded by imagine they have, i image that maybe just a feature like a connect function with a PIN parameter would help me to resolve this issue and will be easier to implement then the callback function, because the callback function will require that the connection function inside Java/Swift be splitted in more steps.

Domvel, 2018-08-22:
@sonnenhohl  Workaround on application: If possible, you can disable the Bluetooth PIN feature and create your own PIN system by a characteristic as above described. There is no difference for the user. You connect to the device and ask the user for a PIN. etc. ... Maybe this is a security problem. But a solution.

sonnenhohl, 2018-08-22:
@Domvel This is a good idea, but at this moment i think it will require extra effort to implement proprietary cryptographic/authentication mechanism in firmware and mobile app and may cause a security problem as you mentioned. Also thinking in terms of future products, integrations with third-party softwares using a standard is easier to maintain.

closed 2022-08-06 by dzonen