don/cordova-plugin-ble-central #628: Ionic-v4 and Native BLE (TypeScript 2.7.2)

smartifier-salomaa, 2018-08-10:
When I'm writing data to BLE device with Ionic-v4/TypeScript v2.7.2, the Ionic Native BLE plugin's write function expects ArrayBuffer. The given example of creating an ArrayBuffer (as data.buffer) does not return ArrayBuffer anymore and gives the following error message:

```
Type ‘ArrayBuffer | SharedArrayBuffer’ is not assignable to type ‘ArrayBuffer’.
Type ‘SharedArrayBuffer’ is not assignable to type ‘ArrayBuffer’.
Types of property ‘[Symbol.toStringTag]’ are incompatible.
Type ‘“SharedArrayBuffer”’ is not assignable to type ‘“ArrayBuffer”’.
```

I’m able to compile the code when I use syntax **data.buffer as ArrayBuffer** but I’m still not able to write anything successfully to BLE device.  Reading works OK. Code was working with older versions (Ionic v3 - plugin v.1.1.4 - TypeScript 2.4.2).

Here is a printout what I'm seeing at Xcode console:
```
To Native Cordova ->  BLE write BLE1198508039 ["options": [CE249DA5-9692-528C-54A5-2ED088FB9D0B, 00abba00-536d-6172-7469-666965720000, 00abba02-536d-6172-7469-666965720000, {
}]]
getData
Looking for 00abba02-536D-6172-7469-666965720000 with properties 8
```

I don't get any reply to my write call (i.e. the promise it not resolved).

untilbit, 2018-08-24:
I have similar issue with ArrayBuffer on write function

Domvel, 2018-08-28:
This can be only fixed in the [ionic native repository](https://github.com/ionic-team/ionic-native/tree/master/src/%40ionic-native/plugins/ble). Because this is a pure javascript cordova plugin.
The issue you described is the effects ionic wrapper.

smartifier-salomaa, 2018-08-30:
OK. I submitted a bug report to Ionic Native repo: https://github.com/ionic-team/ionic-native/issues/2673. Let's hope this is fixed soon.

smartifier-salomaa, 2018-10-05:
Hi @mxdmedia ! Yes. You are correct. Only solution seems to be switching the project from Capacitor back to Cordova. That solved my other problem with Native Audio too.

closed 2018-10-05 by smartifier-salomaa

mxdmedia, 2018-10-05:
@smartifier-salomaa - I dug through Capacitor and found/fixed the issue.  You can take a look at the ionic-capacitor pull request here: https://github.com/ionic-team/capacitor/pull/860  .  Essentially Capacitor was not calling a function that properly encodes ArrayBuf arguments before being passed to the native side.  Once I added in that function call, both my vanilla code and Ionic-native code works great in Capacitor for write operations.  (Note- you still need to make sure your ArrayBuffer is properly typed if using the Ionic-native plugin).

jcesarmobile, 2018-10-09:
This PR also fixes it for Android https://github.com/ionic-team/capacitor/pull/868