don/cordova-plugin-ble-central #632: writeWithoutResponse not working on iOS

mlourenco, 2018-09-07:
Hi,

I'm having some trouble to get my app to work in iOS
The device I am connecting to, requires that I send the message encrypted and split in packages, the message Is build correctly and is working nicely in Android

`"characteristics": [
{
"properties": [
"Read",
"WriteWithoutResponse",
"Notify"
],
"isNotifying": false,
"characteristic": "FFE1",
"service": "FFE0"
}
],
"id": "A700BF1A-EE90-F994-6FE3-307C9FE0DDF4"`

[Log] [BT] writeWithoutResponse("A700BF1A-EE90-F994-6FE3-307C9FE0DDF4", "FFE0", "FFE1",[package-1]: 6c515a2352d57f29cbe34047811a1dfe0965a967) (cordova.js, line 1731)
[Log] [BT] writeWithoutResponse("A700BF1A-EE90-F994-6FE3-307C9FE0DDF4", "FFE0", "FFE1",[package-2]: 6c528d383e3a017d92c2b2e7aa1d833994fc898d) (cordova.js, line 1731)
[Log] [BT] writeWithoutResponse("A700BF1A-EE90-F994-6FE3-307C9FE0DDF4", "FFE0", "FFE1",[package-3]: 6c535a5870993e407d0624d58c1db1089250a3a9) (cordova.js, line 1731)
[Log] [BT] writeWithoutResponse("A700BF1A-EE90-F994-6FE3-307C9FE0DDF4", "FFE0", "FFE1",[package-4]: 6c547950c54c419821af642eed83a1042051a4d4) (cordova.js, line 1731)
[Log] [BT] writeWithoutResponse("A700BF1A-EE90-F994-6FE3-307C9FE0DDF4", "FFE0", "FFE1",[package-5]: 6c55224d01463cfbbffc240625762e241f) (cordova.js, line 1731)
[Log] package successfull written - result: null (cordova.js, line 1731, x5)
(this means all packages were written "x5")

the data is only in hexa on logs, I'm converting it to an ArrayBuffer

before sending the message, I turn On the notifications, that I receive in packages as well. And there's a led light on the device that will be turned on when the message is written, again, on android all is working properly, but on iOS no notify returned and the light never gets on.

The iOS I am doing the tests is version 11.4

Not sure if this is a plugin issue or not, any ideas to what could be the problem? Or how can I help you help me?

mlourenco, 2018-09-23:
seems that the package size should be different, 20 bytes on Android and 60 bytes on iOS, problem solve, not related to the plugin

closed 2018-09-23 by mlourenco

Ariaal, 2019-07-08:
hello ! I get the same question.But i send 60byte every time ,it show me did not wirte well in my device but in my code show me it sent? may I know you code ?

mlourenco, 2019-07-08:
> hello ! I get the same question.But i send 60byte every time ,it show me did not wirte well in my device but in my code show me it sent? may I know you code ?

Hello, hope this helps you, the only thing I did was to change the package size for each device:

`_sendDataInPackages: function (device_id, data, callback, errorCallback) {
          //console.log('Data to send: %s', data);
          var packageDataSize = 18 //+2bytes header (20 total)
          if (window.device.platform !== "Android") {
            packageDataSize = 58; // (60 total)
          }
          var successPackages = 0;
          var npacks = Math.ceil(data.length / 2 / packageDataSize);
          var packages = [];
          var val = "";
          var writeErrorSent = false;
          for (var i = 0; i < npacks; i++) {
            val = data.substr(i * packageDataSize * 2, packageDataSize * 2);
            packages.push("6c" + npacks + (i + 1) + val);
          }
          for (var i2 = 0; i2 < packages.length; i2++) {
            var data2 = new Uint8Array(packages[i2].length / 2);
            for (var j = 0, k = 0; j < packages[i2].length; j += 2, k++) {
              data2[k] = parseInt(packages[i2].substr(j, 2), 16);
            }
            //console.log('%c [BT] write("%s", "%s", "%s",[package-%s]: %s)', 'color:blue;', device_id, this.service_uuid, this.characteristic_uuid_write, i2 + 1, packages[i2]);
            window.ble.writeWithoutResponse(
              device_id,
              this.service_uuid,
              this.characteristic_uuid_write,
              data2.buffer,
              function () {
                //console.log('package successfull writen - result: %o', result);
                successPackages++;
                if (successPackages === packages.length && callback) {
                  callback();
                }
              },
              function (error) {
                console.log('package not writen - error: %o', error);
                if (errorCallback && !writeErrorSent) {
                  writeErrorSent = true;
                  errorCallback({ "message": "[ble] write error", "error": error });
                }
                this.disconnect();
              }.bind(this)
            );
          }
        }`

Ariaal, 2019-07-09:
@mlourenco thank you so much! I want to know if your device has obtained your sent data. I am currently using ios12.3. I sent the data successfully, but the device has no data to pass in, and the connection is unstable, often suddenly disconnected.
Good luck!