don/cordova-plugin-ble-central #639: Why dont you merge the DFU features from "fxe-gear/cordova-plugin-ble-central"?

LabDeve, 2018-10-01:
Why dont you merge the DFU features from "fxe-gear/cordova-plugin-ble-central"?

EarthLord, 2019-01-23:
Please do so. It'll be helpful for many of us.

don, 2019-10-18:
The idea was to keep them separate since that is nRF specific. A better solution might be to move the DFU to a separate plugin that works with this code. @zabojad was working on some plugin hooks which would probably help here.

If someone makes clean pull request I'll probably merge it in. If you're creating a pull request, it would be good to make the Nordic dependencies optional. I like this idea, but I'm not willing to spend time implementing it.

/cc @tomasbedrich

closed 2019-10-18 by don

tomasbedrich, 2020-01-27:
I also like the idea, but I don't work with Cordova anymore, therefore I cannot serve. 🤷‍♂