don/cordova-plugin-ble-central #640: Scan callback delays

FLATT-Dev, 2018-10-09:
I have a custom hardware that transmits undirected advertising packets at 2 second intervals 
In the mobile app I use scanWithOptions API to allow duplicates . 
This works  just fine - in the app I see packets delivered to the scan callback at approximately 2 second intervals most of the time. However occasionally there is up to 5 second delay between the packets.
Here is an example from my log ( TS = Timestamp )
19:18:2:929 Parsing packet 1022, TS = 1956.00
19:18:4:880 Parsing packet 1023, TS = 1951.00
19:18:6:836 Parsing packet 1024, TS = 1956.00
**19:18:10:742 Parsing packet 1025, TS = 3906.00**
**19:18:10:742 Packet TS > 2000!!!**
19:18:12:695 Parsing packet 1026, TS = 1953.00
19:18:14:650 Parsing packet 1027, TS = 1955.00
**19:18:18:554 Parsing packet 1028, TS = 3904.00
**19:18:18:555 Packet TS > 2000!!!****
etc.
The problem gets worse when the hardware starts transmitting at a faster rate (i.e 500 ms)
Then I see  a 3-5 second delay in the log and then the packets arrive in rapid succession at 10-50 ms intervals
This happens on Android (don't know about iOS yet)
Any ideas? What can be done to improve this ?
Thanks
Andy

FLATT-Dev, 2018-10-10:
Update
On Android I added a debug statement to  public void onLeScan  callback and rebuild my app
With only 1 device broadcasting I see delays at this level as well
I assume the packets are doubled because the device is configured to send a scan response.
The expected interval is 2 seconds and that's what I see most of the time.
However occasionally there are delays
This is on Galaxy 3S but is reported on other phones as well
------------------------------------------------------------------
10-09 16:53:58.385  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:53:58.390  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
2 seconds
10-09 16:54:00.335  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:00.340  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
2 seconds
10-09 16:54:02.290  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:02.300  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
2 seconds
10-09 16:54:04.245  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:04.250  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
6 second gap
**10-09 16:54:10.105  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:10.115  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B**
Then a 4 second gap
**10-09 16:54:14.015  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:14.020  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B**
Less than 2 seconds
**10-09 16:54:15.960  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:15.970  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B**
Back to 2 seconds
10-09 16:54:17.915  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:17.920  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
10-09 16:54:19.870  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B
0-09 16:54:19.875  7683  7683 D BLEPlugin: Got packet for 00:A0:50:3F:17:0B

don, 2019-10-18:
This is all stuff that Android is controlling. You could try editing the plugin and adjusting [requestConnectionPriority](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestConnectionPriority(int)).

closed 2019-10-18 by don