don/cordova-plugin-ble-central #642: Scan don't show the Bluetooth peripheral name

Dudusk, 2018-10-29:
Hi !

Good job for your plugin. Work very well but when i scan my bluetooth peripheral, i get no name.

This is my code: 
`
this.Ble.scan([], 5).subscribe(
  device => {
    this.ngZone.run(() => {
      this.devices.push(device);
      alert(JSON.stringify(device))
    });
  },
  error => {
    console.error(error);
  }
);
`

My alert show me : 
`{"id":"MAC:ADDRESS", "advertising":{}, "rssi":-79}`

So, has anyone has this problem ? Do i something wrong ?

Thanks a lot for your answers.
Bye !

miquelferrerllompart, 2018-11-26:
Hi!

We have the same problem here.
After we connect to a device some times, when we try to find it again the device don't show their name.

Our package is:

`"@ionic-native/ble": "~4.6.0"`

Thanks.

woodz-, 2019-01-06:
@Dudusk: it looks like you are mixing things up here. Please show the full code to help out. Some thoughts:

- why are you specifying `Ble` and not `ble`? Is it due to encapsulation?
- `subscribe` is function term coming from the [BluetoothSerial](https://github.com/don/BluetoothSerial) plugin. Why do you use it here on ble? Are you again encapsulating stuff here?
 A more or less counterpart for ble would be the notification principle `ble.startNotification`, `ble.stopNotification` even available for states
- blindly listening to any ble device coming from a scan feels also a bit odd

droghio, 2019-01-12:
Might be a bit of a hack but have you tried looking at the advertising field? For Android this is the raw advertising bytes and if needed will contain the name. Similarly on iOS you can access the raw name here as well. Could help determine if the correct data is being sent from the BLE device.

I've run into name caching problems on iOS and Android before, but using a custom parser on the advertising field usually works. Keep in mind I have seen instances when first starting advertising that I get two advertisements from a device, usually the first is partially invalid but the second and on is good. I suspect the issue is in my BLE device and not the plugin but thought I'd give the warning.

don, 2019-10-18:
This plugin just passes along the data from the underlying OS. Sometime the device name is in the advertising packet, other times it's in the scan response packet. In either of these cases it is 0x09 in the advertising data. Android tends to use the this value.

iOS tends to use the Generic Access Service 0x1800 and the Device Name Characteristic 0x2A00 to get the name that is displayed. Often on iOS you'll see old cached values until connecting.

closed 2019-10-18 by don