don/cordova-plugin-ble-central #643: Scanning before connecting

fernandoSynetic, 2018-11-01:
The documentation indicates that it is possible to connect without making a previous scan. It does not work for me.

I do:
ble.connect("FF:5D:1A:36:AF:07", function(data){alert (JSON.stringify(data, null, 2));},function(){alert ("No connect");});

If I do a previous scan, it works.
Any ideas?
Thank you

cairinmichie, 2018-11-15:
From my experience, you need to scan before using `ble.connect`. iOS doesn't support connecting to devices without scanning for them first. If you want to do this on android you can use `ble.autoConnect`.

fernandoSynetic, 2018-11-19:
thanks for the reply.
I am using android. I have also tried autoConnect but it does not work either.

Has this functionality ever worked?
Thank you.

Domvel, 2018-12-05:
I also wrapped the scan and connect methods in an own async method to scan before connect. On Android you can connect without a scan. But in my case, the device name I want, also comes with a scan. If I only connect without knowing the meta data but the MAC ID, the name does not appear in the discovered data of the connect method. So a scan before connect is recommended.

don, 2019-10-18:
@fernandoSynetic I'm not sure why this isn't working for you. I suggest stepping through the debugger with Android Studio to see what's happening

https://github.com/don/cordova-plugin-ble-central#scanning-before-connecting

closed 2019-10-18 by don