don/cordova-plugin-ble-central #644: BLE connectivity breaks down after a few minutes

mcgengineer, 2018-11-09:
Hello,
I'm using a basically unmodified version of your code to make a Cordova project that's intended to read Feather 32u4 with simple gyro data transmission for the data from my Adafruit LSM9DS1.
Generally speaking, the code runs correctly for several minutes. After that, no BLE connectivity error seems to be thrown in either the console or the app; however, the gyro data stops transmitting from the Feather to the app (just stops updating for a while, and then starts sending random numbers). It can still be read over Serial and the disconnect button no longer works properly, so I think it's an issue with BLE connectivity.
I've spent a few months migrating my project from another Cordova-based project (to communicate with a different Arduino-type microcontroller using BLE) to this framework in order to keep using Cordova, but if I can't resolve this issue, I'll have to rewrite in Swift (and learn Swift!) since the Adafruit native Swift app doesn't seem to have this BLE connectivity problem. Can you possibly help?
(I put more details about what changes I made to this code and the code I uploaded to the Feather here. However, since the gyro data is transmitting correctly over Serial, it looks like the issue is actually with the Cordova project - https://forums.adafruit.com/viewtopic.php?f=57&t=143341)
Thank you!
- Miri

woodz-, 2019-01-06:
I would suggest to try on Android. If you have the comfort of OS independence you even have the comfort to test on different platforms.
I had a similar problem a view month ago, but the other way around. My application under Android slowed down after a few seconds, while under IOS it worked pretty stable. At the end, changing to another ble peripheral was the only solution to support both platforms.
My experience is that ble on IOS is much more stable and free of bugs than the Android's stack.

don, 2019-10-18:
Closing old issue.

closed 2019-10-18 by don