don/cordova-plugin-ble-central #645: Range decreases over time. (Android only)

cairinmichie, 2018-11-15:
I have an implementation of this pluign that interfaces with the ESP32.

Initially when the plugin first starts on a fresh install, the range is great on both android and iOS. I get roughly 50m easily on both.

However, after a few days of intermittent use on android the range when scanning seems to drop off dramatically. Such that I need to be within 10m of a device to scan it. This doesn't happen on iOS, and the device can be found from the same range as before.

The reason I think this is a problem with the plugin and not my ESP32 implementation is because:

- The plugin continues to find devices at range on iOS.
- If I then use `ble.autoConnect` the android device that previously couldn't find a device, will magically connect to it at range. And if I then scan for a device after using `ble.autoConnect` I can see the device at range again.

I would just use `ble.autoConnect`, however that would require separate implementations on android/iOS which isn't ideal. More information on this given at #620.

Is there some sort of caching that happens with found devices that I need to be aware of that would somehow 'clog' the ble scan? If so, is there a way to clear this/ prevent this?

My scanning code looks like this:

```
var services = ['00FF']
ble.startScan(services, (peripheral) => {
    // Connect to device code here.
}, (err) => { console.error('ble.startScan error: ', err) })
```

droghio, 2019-02-07:
Do you have the same problem with other ble scanners on your same Android device?

cairinmichie, 2019-02-07:
@droghio Not as far as I'm aware. Although I haven't tested nearly as much with other apps.

It happens with quite a variety of android devices. We've had the problem with Huawei, LG and the Nexus 6p(Huawei manufactured) devices that I'm aware of.

don, 2019-10-18:
This sounds like an Android problem. Closing since it's an old issue.

closed 2019-10-18 by don