don/cordova-plugin-ble-central #649: ble.isConnected returning "OK" when not connected

sonnenhohl, 2018-12-05:
Im having a strange behaviour where the function ble.isConnected is returning "OK" when:

1) Connect to bluetooth device
2) Exchange some data
3) Turn of bluetooth on android
4) ble.isConnected keeps returning "OK" even with bluetooth off.

sonnenhohl, 2018-12-05:
Another way to do it is instead of turning of the bluetooth on android turn of the physical device, one of my phones just bugs with this second approach the other with both of them.

motla, 2018-12-06:
Apparently in the documentation there is a mismatch between the argument description and the quick example:

> ### Parameters
> **device_id:** UUID or MAC address of the peripheral
> **success:** Success callback function that is invoked with a boolean for connected status.
> **failure:** Error callback function, invoked when error occurs. [optional]
> ### Quick Example
> ```
> ble.isConnected(
>     'FFCA0B09-CB1D-4DC0-A1EF-31AFD3EDFB53',
>     function() {
>         console.log("Peripheral is connected");
>     },
>     function() {
>         console.log("Peripheral is *not* connected");
>     }
> );
> ```

Maybe it should be:

### Quick Example
```
ble.isConnected(
    'FFCA0B09-CB1D-4DC0-A1EF-31AFD3EDFB53',
    function(connected) {
        if(connected) console.log("Peripheral is connected");
        else console.log("Peripheral is *not* connected");
    },
    function() {
        console.log("An error occured");
    }
);
```

If it solves the problem, please consider a push request of the documentation...

cactrot, 2019-02-05:
In the Android and IOS implementations it appears to call the success callback without the Boolean argument. 

https://github.com/don/cordova-plugin-ble-central/blob/ac55995dc9e9d19f231c13e6bbfb9ba24b076480/src/android/BLECentralPlugin.java#L249

https://github.com/don/cordova-plugin-ble-central/blob/ac55995dc9e9d19f231c13e6bbfb9ba24b076480/src/ios/BLECentralPlugin.m#L337

And the example code matches the description body. The parameter description is most likely incorrect.

I just came across similar behavior but I was always getting null for the argument.

don, 2019-10-18:
The [example](https://github.com/don/cordova-plugin-ble-central#quick-example-7) for [isConnected](https://github.com/don/cordova-plugin-ble-central#isconnected) is correct.

Especially on Android disconnects are not immediate. The phone doesn't know the device is disconnected. I'd suggest handing disconnects by adding disconnectCallback function on [connect](https://github.com/don/cordova-plugin-ble-central#connect). The plugin will notify you when the device disconnects.

closed 2019-10-18 by don

panuhorsmalahti, 2020-02-25:
The documentation is still incorrect: "Success callback function that is invoked with a boolean for connected status.". The parameter invoked is a string "OK", not a boolean.