don/cordova-plugin-ble-central #652: Receiving only one notification on Android but all of them on iOS

defc0n1, 2018-12-14:
On Android I use ble.requestMtu to increase MTU since the peripheral is sending more data. On iOS I use the same code without requestMtu and I am receiving all notifications as intended. I do however not receive more then one notification on Android which always has the same content: {"0":0,"1":0,"2":128,"3":0}

I tested with LightBlue and nRF Connect and with the same process it works on Android as well.

untilbit, 2019-02-27:
The plugin only sends one notification/callback when scanning.

@don Could you improve this? thanks in advance

don, 2019-10-18:
I don't recommend using requestMtu. This issue is really old, so closing for now. Re-open if it's still a problem.

closed 2019-10-18 by don